chipsalliance/firrtl #1513: Some type cannot be mapped but can fit in to Seq.

sequencer, 2020-04-13:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
In currently implementations of IRs, we use `Seq` to store sub-field.
and define `map` or `foreach` functions map on that field
e.g.:
```
case class Module(info: Info, name: String, ports: Seq[Port], body: Statement) extends DefModule
```
and we can use 
```
val someModule: Module
someModule.ports.map(???)
someModule.ports.foreach(???)
```
However a `scala.collection.immutable.Vector` will also fit the `Seq[Port]` type constraint.
which will break the port map function.(only yield first element)
As the example in https://github.com/sequencer/diplomatictester/blob/99cc6c1e87fe76756df1851f4f531834931eaf0b/src/firrtl/DutIOTransform.scala#L73-L75
I fixed this with `topPorts.values.toArray.toSeq` to get a `scala.collection.mutable.WrappedArray` to fix this bug.

### What is the expected behavior?
any type of `Seq[Port]` should use API like `map` `foreach`

### Steps to Reproduce
Error code line:
https://github.com/sequencer/diplomatictester/blob/99cc6c1e87fe76756df1851f4f531834931eaf0b/src/firrtl/DutIOTransform.scala#L73-L75
This can be tested with
```
git clone https://github.com/sequencer/rocket-playground/
cd rocket-playground
git submodule update --init
git checkout 78e22787508bdeef88e51289321664a9154813f5
mill diplomatictester.tests.runMain "diplomatictester.tests.DutIOTester"
```

### Your environment
- Chisel:  a39d0e1
- Firrtl:  eb52a5c
- Java: OpenJDK 11
- OS: Arch Linux

albert-magyar, 2020-04-28:
TLDR: you are introducing lazy evaluation into your code with `mapValues`, which is breaking your implicit expectation of strict side effect visibility.

The problem is not the use of `Vector`, but that you make a module with a `Stream` for the `ports` field. In fact, a `Vector` of ports [works just fine](https://scastie.scala-lang.org/8wDViQCzT5mB3IKsNI6HPw).

Nevertheless, a `Stream` can be mapped and will in fact produce the expected results for the entire stream; however, the mapped 'tail' is **lazily evaluated**, which means that side-effect-having functions will not show their side effects immediately after the `map` call (hence the problem with the prints).

One way to make this more robust is to use `foreach` when it's necessary to rely on side effects in IR traversals. Unlike `map`, this forces the immediate materialization of the entire `Stream` based on the assumption that `foreach` will be combined with side effects.

Back to the concrete issue: why do we get a lazy `Stream` when Scala is generally "strict-by-default?"

By calling `mapValues` on a Map, you can get lazy evaluation! Furthermore, there are also other reasons to avoid using `values`. Try the following replacements to restore eager behavior for your side-effect-having code:

- `myMap.map { case (k, v) => k -> f(v) }` instead of `myMap.mapValues(f(_))`
- `myMap.map { case (k, v) => v }` instead of `myMap.values`

Unfortunately, there isn't a convenient upper type bound to force a `Seq` to have an eager `map`. While the `IndexedSeq` trait does require eager element evaluation, it precludes linked lists, which are the default concrete type of the `Seq` factory! The "best" way to avoid these issues is to avoid relying on `map` to cause side effects before traversing the output of the `map`, and to instead use `foreach`, but in cases where this is violated, it is critical to avoid introducing laziness with `<map>.values`.

closed 2020-04-28 by albert-magyar

jackkoenig, 2020-04-28:
I ran into this a long time ago and started looking for a solution: https://stackoverflow.com/questions/52524815/enforce-strict-collections

But that approach would be very painful to deploy throughout the codebase, so the pitfall remains. Notably, the Scala devs deprecated `.mapValues` in Scala 2.13 because of this.

Something else we could consider is moving to concrete types instead of using `Seq` in the datastructure, maybe `Vector` which was made much faster in 2.13.

albert-magyar, 2020-04-28:
> I ran into this a long time ago and started looking for a solution: https://stackoverflow.com/questions/52524815/enforce-strict-collections
>
> But that approach would be very painful to deploy [...]

I think it's for the best that we don't have upper bounds like that -- it would solve some instances of the "lazy side effect issues" while allowing others within the body of a transform.

We should maybe have a doc listing guidelines like a) use `foreach` when appropriate and b) ban `mapValues` :)

sequencer, 2020-05-02:
Thank you, learned a lot!
I found In 2.12, my code directly generated `Stream`. but in 2.13, it is a normal `Seq`. I think in the future(may be 2.13), we can constrict type tighter to get rid of this situation.