chipsalliance/firrtl #1554: Prefix FIRRTL-emitted macros with FIRRTL_

debs-sifive, 2020-05-01:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Prefix Prefix FIRRTL-emitted macros with `FIRRTL_`. This could help if users wanted to define a macro such as `FIRRTL_RANDOM` differently for their FIRRTL-generated code than for the macro `RANDOM` elsewhere in their code.

Supporting this may have to include a way to default the values of macros such `FIRRTL_RANDOM` to the non-prefixed `RANDOM`.

### Type of Feature

- code refactoring

### Related Features

#1550 is an example of this.

### External Information

There is some discussion going on in #1550.

mwachs5, 2020-05-01:
Ah, i thought in #1550 you would just do it for the new macros you were adding... seems like this is a bigger breaking change?

debs-sifive, 2020-05-01:
Yes, that is correct. #1550 only prefixes the newly added macros (`FIRRTL_BEFORE_INITIAL` and `FIRRTL_AFTER_INITIAL`).