chipsalliance/firrtl #1562: The specification appears to be silent on what precisely constitutes indentation

boqwxp, 2020-05-06:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

[The spec](https://github.com/freechipsproject/firrtl/blob/master/spec/spec.pdf) mentions "indent" twice, both on page 58.  Specifically, it says:

> Block structuring is indicated using indentation. Statements are com-
> bined into statement groups by surrounding them with parenthesis. A colon
> at the end of a line will automatically surround the next indented region
> with parenthesis and thus create a statement group.

This leaves the reader (me) unclear what exactly constitutes indentation.  Is a '\t' indentation?  A single space?  Two spaces?  Four?  Can I mix and match spaces and tabs?

### What is the expected behavior?

The spec will precisely define what is acceptable as indentation.

### Steps to Reproduce

Open [the spec](https://github.com/freechipsproject/firrtl/blob/master/spec/spec.pdf).  Search for "indent".

### Your environment

N/A

### External Information

See [this Yosys issue](https://github.com/YosysHQ/yosys/issues/2004).

seldridge, 2020-05-06:
Pinging @azidar, @jackkoenig, and @albert-magyar to weigh in.

Spec needs to have the final say on this, but the *current implementation* works as follows:

Spaces only, no tabs. Number of spaces can be anything, but just needs to be consistent (indentation and dedentation need to match).

E.g., the FIRRTL compiler will accept 8 spaces for indenting modules and then 4 additional spaces of indentation in one module and 2 spaces of indentation in the second:

```
circuit Foo:
        module Foo:
            skip
        module Bar:
         skip
```

However, the compiler will not accept mixing indent levels (4 spaces for first module, 2 spaces for the second) because the indentation doesn't match:
```
circuit Foo:
    module Foo:
      skip
  module Bar:
    skip
```

There was some discussion about this in the 2020-03-16 dev meeting related to the original request to allow tabs, #486. The conclusion was to disallow tabs because there wasn't an answer on how to handle mixed tabs vs. spaces. The library we use for identation/dedentation in antlr4 didn't have a way to say "tabs only" or "spaces only" and enabling tabs would result in allowing Python-style tab/space mixing. We didn't want this and have been consuming/emitting spaces all along so we stuck with that. Tab support, #1435, was then closed.

albert-magyar, 2020-05-06:
We add something verbose like:

In FIRRTL, indentation is significant and may consist of spaces only. The number of spaces appearing before a line of FIRRTL IR is used to establish its _indent level_. The indent level of the top-level Circuit declaration must be zero. Certain constructs (Circuit, Module, Conditionally) create a new sub-context; all IR nodes that are direct children of such a construct must be indented by the same non-zero number of spaces from the indent level of the parent construct. Together, these indented statements form a statement group [ADD REF]. Since Conditionally statements may be nested, it is possible to create a hierarchy of indent levels, each with its own number of preceding spaces that must be larger than its parent’s and consistent among all direct child statements (those that are not children of an even deeper Conditionally).

The indent used on the first line that is part of the new sub-context establishes the number of spaces constituting the indent level of that construct, which is one higher than the parent; any following line that is a direct child of the same sub-context (and not a child of a new, deeper sub-context) must be indented by an identical number of spaces, or a syntax error will result. To end the sub-context and return to the context of the parent, a line must return to the indent level of the parent, which requires it to be indented by an identical number of spaces as the parent.

As a concrete guide, a few consequences of these rules are summarized below:
- The top-level circuit declaration must not be indented from the left margin.
- The declarations beginning all modules must be indented by the same number of spaces.
- Within a module, all port declarations and statements that are not themselves children of statements must be indented by the same number of spaces. Since the parent context is the module, the number of spaces comprising this indent level is specific to each module.
- All child statements of the statement group within the `when` or `else` case of a Conditionally statement must be indented by the same number of spaces, unless they are children of a statement group of a deeper Conditionally statement that would establish its own, strictly deeper indent level.
- Each `when` and each `else` may have a distinct number of spaces in its indent level, as long as it satisfies the constraint that its indent level is deeper than that of its parent.

PR #1565 (2020-05-06): Clarify spec definition of indentation and when/else indentation
Updates the spec to reflect the current FIRRTL compiler implementation of indentation:

1. Indentation consists only of spaces, no tabs
2. Indentation is locally defined in a block and not global (you can indent the when condition by two spaces and the else condition by ten spaces)
3. Any mention of the extinct parentheses is removed. The when/else syntactic shorthand section is updated related to indenting or not-indenting when/else conditions.

This is all based on the extant antlr4.gr grammar and running FIRRTL IR snippets thr [... cut]
closed 2020-05-06 by seldridge