chipsalliance/firrtl #1586: upgrade antlr4 to version 4.8

itviewer, 2020-05-11:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

upgrade antlr4 to version 4.8

### Type of Feature

### Related Features

### External Information
https://github.com/antlr/antlr4/releases
https://github.com/antlr/antlr4/commits/master/runtime/Java

I want to add a new features to the [Intellij plugin](https://github.com/easysoc/easysoc-firrtl), which depends on the firrtl jar package. However my plugin depends on antlr 4.8 and the firrtl package depends on the very old 4.7.1.
I tried to downgrade the dependency of the Intellij plugin from 4.8 to 4.7.1, but it occasionally crashed. Then I tried to upgrade the antlr version that the firrtl package depends on to 4.8. 
So far, I have not encountered any problems.

In the link above, we can find the upgrade content. There should be few changes to the java target. Currently, I do n’t think upgrading antlr4 to the latest version will cause any problems. 
Of course, maybe this requires more testing.

itviewer, 2020-11-02:
@sequencer 
Could you help evaluate this issue?

sequencer, 2020-11-02:
Sure I'll put my hand on it today.

azidar, 2020-11-02:
Related: https://github.com/freechipsproject/firrtl/pull/1849

sequencer, 2020-11-03:
Hi, @itviewer.
Here is the discussion result in the dev meeting.
We should upgrade this version, but need to check binary compatibility.
1. If upgrade doesn’t break binary compatibility, directly merge #1849
2. If breaks, need to wait until FIRRTL 1.5 to get it merged.
I’ll do the binary check today.

itviewer, 2020-11-03:
I used some simple circuits for testing. After upgrading antlr to 4.8, the Firrtl generated by the circuit is no different.

In addition, #1849 need to add `build.sbt` support.

sequencer, 2020-11-03:
Sure I'll rebase your #1587 to master and do the binary check.

sequencer, 2020-11-03:
Binary check result(Open with browser): [compat_report.txt](https://github.com/freechipsproject/firrtl/files/5482549/compat_report.txt).
checked with japi-compliance-checker 2.4. need  @jackkoenig review to decide whether to upgrade in this version or not.

jackkoenig, 2020-11-03:
(From the report)

#### High Severity

class org.antlr.v4.runtime.atn.LexerATNSimulator
 Change | Effect
-- | -- |
Field match_calls of type int has been removed from this class. | A client program may be interrupted by NoSuchFieldError exception.

### Low Severity

class org.antlr.v4.runtime.RuntimeMetaData

  Change | Effect
-- | -- |
Value of final field VERSION of type java.lang.String has been changed from "4.7.1" to "4.8". | Old value of the field will be inlined to the client code at compile-time and will be used instead of a new one.

jackkoenig, 2020-11-03:
The `High Severity` issue is a real binary incompatible change. It's only 1 such issue so I think it's probably okay to go ahead and merge this. It might be worth compiling FIRRTL against 4.7.1 and running it against 4.8 and see if it hits this issue.

sequencer, 2020-11-03:
I didn't get:
> compiling FIRRTL against 4.7.1 and running it against 4.8.

Does this mean Java code generated by antlr4 4.7.1, and use `org.antlr:antlr4-runtime:4.8` to run firrtl?

sequencer, 2020-11-03:
since sbt is supported by plugin, I just test with mill by 
```diff
diff --git a/build.sc b/build.sc
index 5ba49d12..732b9c44 100644
--- a/build.sc
+++ b/build.sc
@@ -45,7 +45,7 @@ class firrtlCrossModule(val crossScalaVersion: String) extends CrossSbtModule wi
       ivy"net.jcazevedo::moultingyaml:0.4.2",
       ivy"org.json4s::json4s-native:3.6.9",
       ivy"org.apache.commons:commons-text:1.8",
-      ivy"org.antlr:antlr4-runtime:$antlr4Version",
+      ivy"org.antlr:antlr4-runtime:4.8",
       ivy"com.google.protobuf:protobuf-java:$protocVersion"
     ) ++ {
       if (majorVersion > 12)
@@ -101,7 +101,7 @@ class firrtlCrossModule(val crossScalaVersion: String) extends CrossSbtModule wi
   }

   def downloadAntlr4Jar = T.persistent {
-    Util.download(s"https://www.antlr.org/download/antlr-$antlr4Version-complete.jar")
+    Util.download(s"https://www.antlr.org/download/antlr-4.7.1-complete.jar")
   }

   def generatedAntlr4Source = T.sources {
```
It passed all test. I think it's OK to merge?

jackkoenig, 2020-11-03:
That's still not quite right (although close), you want only the runtime ivydep to differ (not just the ivyDeps which includes compile time). I did it manually by taking the full FIRRTL classpath as resolved by coursier:
```bash
$ coursier fetch edu.berkeley.cs:firrtl_2.12:1.4.0 > classpath
```
Then editing `antlr4-runtime` to `4.8`, then you can run build a classpath from that:
```bash
$ CP=$(cat classpath | paste -sd ":" -)
```
Then running FIRRTL:
```bash
$ java -cp $CP firrtl.stage.FirrtlMain -i regress/FPU.fir -o out -X verilog
```
It works which is good, but does print a bunch of warnings
```
ANTLR Tool version 4.7.1 used for code generation does not match the current runtime version 4.8ANTLR Runtime version 4.7.1 used for parser compilation does not match the current runtime version 4.8ANTLR Tool version 4.7.1 used for code generation does not match the current runtime
version 4.8ANTLR Runtime version 4.7.1 used for parser compilation does not match the current runtime version 4.8
```

jackkoenig, 2020-11-03:
@itviewer can you try the plugin against ANTLR 4.7.2? It does have some fixes for the Java target.

itviewer, 2020-11-04:
@sequencer @jackkoenig 
I can't  really understand what binary compatibility means here?
Every time, we use Antlr4 to generated the lexer/parser code, **that** code contains the detection of the current runtime library.
The Antlr used to generate the lexer/parser should **exact** match the runtime library version which the lexer/parser use, that is:
- if we use the lexer/parser generated by Antlr 4.8, we should rely on the runtime library of version 4.8
- if we use the lexer/parser generated by Antlr 4.7.1, we should rely on the runtime library of version 4.7.1
- if we use the lexer/parser generated by Antlr 4.7.2, we should rely on the runtime library of version 4.7.2

The version number used for version detection is defined in [RuntimeMetaData.java](https://github.com/antlr/antlr4/blob/master/runtime/Java/src/org/antlr/v4/runtime/RuntimeMetaData.java). from the [history commits ](https://github.com/antlr/antlr4/commits/master/runtime/Java/src/org/antlr/v4/runtime/RuntimeMetaData.java), we can see that 4.7.2 is a different version from 4.7.1. So, if we use Antlr 4.7.1 to generate the lexer/parser and use the runtime library of version 4.7.2, the warning will **also** appear. 
(We should not use one version of Antlr to generate the lexer/parser, but use another version of the runtime library)
and, from the [java target history commits](https://github.com/antlr/antlr4/commits/master/runtime/Java), we can see that the **4.8** version has **fewer** changes compared to the **4.7.2** version.

It seems we’d better either directly upgrade to version 4.8 or stay at 4.7.1

sequencer, 2020-11-04:
> That's still not quite right (although close), you want only the runtime ivydep to differ (not just the ivyDeps which includes compile time). I did it manually by taking the full FIRRTL classpath as resolved by coursier:
> 
> ```shell
> $ coursier fetch edu.berkeley.cs:firrtl_2.12:1.4.0 > classpath
> ```
> 
> Then editing `antlr4-runtime` to `4.8`, then you can run build a classpath from that:
> 
> ```shell
> $ CP=$(cat classpath | paste -sd ":" -)
> ```
> 
> Then running FIRRTL:
> 
> ```shell
> $ java -cp $CP firrtl.stage.FirrtlMain -i regress/FPU.fir -o out -X verilog
> ```
> 
> It works which is good, but does print a bunch of warnings which is pretty annoying.
> 
> ```
> ANTLR Tool version 4.7.1 used for code generation does not match the current runtime version 4.8ANTLR Runtime version 4.7.1 used for parser compilation does not match the current runtime version 4.8ANTLR Tool version 4.7.1 used for code generation does not match the current runtime
> version 4.8ANTLR Runtime version 4.7.1 used for parser compilation does not match the current runtime version 4.8
> ```

I wonder in which case this will affect to our users?
Something like use is depending on FIRRTL 1.4 with ivy, and we upgrade the 1.4 binary in the ivy?
If user redownload this in ivy, and used antlr4 related API, will cause user failing?

itviewer, 2020-11-04:
> and used antlr4 related API, will cause user failing

Firrtl already has an IR representation. antlr4 related things are really very low-level, and people who use the Firrtl library basically do not need to use antlr4 directly.

jackkoenig, 2020-11-04:
@itviewer binary compatibility in this context means that if someone compiles their code against FIRRTL v1.4.0 then it should work even if they use FIRRTL v1.4.1 or v1.4.7 at runtime. This is most important for library writers where a user may be using a library but then pick a different version of Chisel or FIRRTL than the library was compiled against (for example: https://github.com/tdb-alcorn/chisel-formal/issues/9). Maintaining binary compatibility means we cannot bump any of our dependencies in a binary incompatible way because it still affects our upstream users. Note that I'm only referring to _within_ major release branches, we can of course bump on master.

> (Even if the API has not changed, it doesn't mean that the logic function has not changed, we should not use one version of Antlr to generate the lexer/parser, but use another version of the runtime library)

This is the purpose of versioning. It is absolutely safe to compile your code against one version of FIRRTL and run against a newer one with the same major version number. The implication is that this should be true for ANTLR 4.7.1 vs. 4.7.2, but there are complaints in the ANTLR repo about breakages between 4.7 and 4.7.1 so perhaps this assumption is wrong.

> So, if we use Antlr 4.7.1 to generate the lexer/parser and use the runtime library of version 4.7.2, the warning will also appear.

Not true, I did the same thing I described above but using ANTLR 4.7.2 instead of 4.8 and there is no warning printed. Your point is well taken that there are a lot of changes so perhaps we shouldn't bump to 4.7.2 on the FIRRTL 1.4.x branch.

With all of this in mind. I am of the opinion that we should bump to ANTLR 4.8 on master but not backport it.

sequencer, 2020-11-05:
I think we can directly bump and not backport, PR only affects `build.sc` and `build.sbt`, which won’t be a conflict in the future.

sequencer, 2020-11-10:
I think this can be merged after #1947

closed 2020-11-16 by mergify[bot]