chipsalliance/firrtl #1621: Infertype didn't infer reference type.

sequencer, 2020-05-14:
When I'm working on the #1603, I found the legacy code doesn't work since `Reference` in a `Statement` has no types. 
I'm not sure about should `Reference` inside `Statement` should have types or not. If not I'll find some other ways to work around.
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### Steps to Reproduce
Under `ammonite`
```scala
import $ivy.`edu.berkeley.cs::firrtl:1.3.0`
val input =
    """circuit Test:
      |  module Test :
      |    input in: UInt<8>
      |    input clk: Clock
      |    input reset: UInt<1>
      |    output out: {a: UInt<8>, b: UInt<8>[2]}
      |    input ana1: Analog<8>
      |    output ana2: Analog<8>
      |
      |    out is invalid
      |    reg r: UInt<8>, clk with:
      |      (reset => (reset, UInt(0)))
      |    node x = r
      |    wire y: UInt<8>
      |    y <= x
      |    out.b[0] <= and(y, asUInt(SInt(-1)))
      |    attach(ana1, ana2)
      |    inst child of Child
      |    out.a <= child.out
      |  module Child:
      |    output out: UInt<8>
      |    out <= UInt(1)
      |""".stripMargin
show(new firrtl.stage.TransformManager(Seq(), firrtl.stage.Forms.Resolved).execute(firrtl.CircuitState(firrtl.Parser.parse(input), firrtl.UnknownForm)).circuit)
```

### What is the current behavior?

```
Circuit(
  ,
  ArrayBuffer(
    Module(
      ,
      "Test",
      ArrayBuffer(
        Port(, "in", Input, UIntType(IntWidth(8))),
        Port(, "clk", Input, ClockType),
        Port(, "reset", Input, UIntType(IntWidth(1))),
        Port(, "out", Output, BundleType(ArrayBuffer(Field("a", Default, UIntType(IntWidth(8))), Field("b", Default, VectorType(UIntType(IntWidth(8)), 2))))),
        Port(, "ana1", Input, AnalogType(IntWidth(8))),
        Port(, "ana2", Output, AnalogType(IntWidth(8)))
      ),
      Block(
        ArrayBuffer(
          IsInvalid(, Reference("out", UnknownType)),
          DefRegister(, "r", UIntType(IntWidth(8)), Reference("clk", UnknownType), Reference("reset", UnknownType), UIntLiteral(0, IntWidth(1))),
          DefNode(, "x", Reference("r", UnknownType)),
          DefWire(, "y", UIntType(IntWidth(8))),
          Connect(, Reference("y", UnknownType), Reference("x", UnknownType)),
          Connect(
            ,
            SubIndex(SubField(Reference("out", UnknownType), "b", UnknownType), 0, UnknownType),
            DoPrim(and, ArrayBuffer(Reference("y", UnknownType), DoPrim(asUInt, ArrayBuffer(SIntLiteral(-1, IntWidth(1))), ArrayBuffer(), UnknownType)), ArrayBuffer(), UnknownType)
          ),
          Attach(, ArrayBuffer(Reference("ana1", UnknownType), Reference("ana2", UnknownType))),
          DefInstance(, "child", "Child"),
          Connect(, SubField(Reference("out", UnknownType), "a", UnknownType), SubField(Reference("child", UnknownType), "out", UnknownType))
        )
      )
    ),
    Module(, "Child", ArrayBuffer(Port(, "out", Output, UIntType(IntWidth(8)))), Block(ArrayBuffer(Connect(, Reference("out", UnknownType), UIntLiteral(1, IntWidth(1))))))
  ),
  "Test"
)
```
### What is the expected behavior?
```
Circuit(
  ,
  ArrayBuffer(
    Module(
      ,
      "Test",
      ArrayBuffer(
        Port(, "in", Input, UIntType(IntWidth(8))),
        Port(, "clk", Input, ClockType),
        Port(, "reset", Input, UIntType(IntWidth(1))),
        Port(, "out", Output, BundleType(ArrayBuffer(Field("a", Default, UIntType(IntWidth(8))), Field("b", Default, VectorType(UIntType(IntWidth(8)), 2))))),
        Port(, "ana1", Input, AnalogType(IntWidth(8))),
        Port(, "ana2", Output, AnalogType(IntWidth(8)))
      ),
      Block(
        ArrayBuffer(
          IsInvalid(, Reference("out", BundleType(ArrayBuffer(Field("a", Default, UIntType(IntWidth(8))), Field("b", Default, VectorType(UIntType(IntWidth(8)), 2)))))),
          DefRegister(, "r", UIntType(IntWidth(8)), Reference("clk", ClockType), Reference("reset", UIntType(IntWidth(1))), UIntLiteral(0, IntWidth(1))),
          DefNode(, "x", Reference("r", UIntType(IntWidth(8)))),
          DefWire(, "y", UIntType(IntWidth(8))),
          Connect(, Reference("y", UIntType(IntWidth(8))), Reference("x", UIntType(IntWidth(8)))),
          Connect(
            ,
            SubIndex(SubField(Reference("out", BundleType(ArrayBuffer(Field("a", Default, UIntType(IntWidth(8))), Field("b", Default, VectorType(UIntType(IntWidth(8)), 2))))), "b", VectorType(UIntType(IntWidth(8)), 2)), 0, UnknownType),
            DoPrim(and, ArrayBuffer(Reference("y", UIntType(IntWidth(8))), DoPrim(asUInt, ArrayBuffer(SIntLiteral(-1, IntWidth(1))), ArrayBuffer(), UnknownType)), ArrayBuffer(), UnknownType)
          ),
          Attach(, ArrayBuffer(Reference("ana1", AnalogType(IntWidth(8))), Reference("ana2", AnalogType(IntWidth(8))))),
          DefInstance(, "child", "Child"),
          Connect(, SubField(Reference("out", BundleType(ArrayBuffer(Field("a", Default, UIntType(IntWidth(8))), Field("b", Default, VectorType(UIntType(IntWidth(8)), 2))))), "a", UIntType(IntWidth(8))), SubField(Reference("child", UnknownType), "out", UIntType(IntWidth(8))))
        )
      )
    ),
    Module(, "Child", ArrayBuffer(Port(, "out", Output, UIntType(IntWidth(8)))), Block(ArrayBuffer(Connect(, Reference("out", UIntType(IntWidth(8))), UIntLiteral(1, IntWidth(1))))))
  ),
  "Test"
)
```

jackkoenig, 2020-05-14:
This is perhaps the biggest wart in the implementation right now. The "real" IR used by FIRRTL is the "Working" IR in WIR.scala. The plan-of-record is to clean this up and merge the IR and the Working IR for the next major release (1.4). See https://github.com/freechipsproject/firrtl/pull/1350. Basically, WRef will become Reference such that all of the fields of WRef will exist in Reference and transforms like InferTypes will operate on Reference.

sequencer, 2020-05-14:
I got it. I'll have a try on `ToWorkingIR`.

sequencer, 2020-05-14:
OK, I'll rebase my work onto #1350 and wait it to solve this issue.

jackkoenig, 2020-05-14:
I wouldn't rebase quite yet. I just wouldn't worry about `Reference` not working, instead focusing on the existing WIR

sequencer, 2020-05-14:
Emmm, I just rebased back to master, and trying to read code to find out why this happened.

albert-magyar, 2020-05-14:
Your issue is that you're providing the constructor parameters to `TransformManager` backwards -- you're actually not running `InferTypes` at all!

[If you look at the API reference](https://www.chisel-lang.org/api/firrtl/1.3.0/firrtl/stage/TransformManager.html#%3Cinit%3E(targets:Seq[firrtl.stage.TransformManager.TransformDependency],currentState:Seq[firrtl.stage.TransformManager.TransformDependency],knownObjects:Set[firrtl.Transform]):firrtl.stage.TransformManager), the list of desired transforms to run (`targets`) is actually the first argument to the constructor. Here, `Resolved` is a convenient shorthand for a list of transforms that leaves you with inferred types, among other things.

The second constructor argument is a list of transforms that _have already been run_ before starting that transform manager -- which means they won't run unless they are invalidated and need to be re-run.

It looks like what you want is something like the the following. Note that `Resolved` is passed as the first parameter to the `TransformManager` constructor.
```scala
(new firrtl.stage.TransformManager(stage.Forms.Resolved)).execute(inputCircuitState)
```

seldridge, 2020-05-14:
Adding to Albert's point: you may want to use `firrtl.stage.transforms.Compiler` instead of a bare `TransformManager`. That `Compiler` will add necessary wrappers around the determined transform order for: (1) running each transform's `prepare` method, (2) updating annotations, and (3) catching exceptions.

sequencer, 2020-05-14:
Oh...... Thank you @albert-magyar @seldridge, My fault, I just found `ToWorkingIR` didn't execuated...
And found your reply.

closed 2020-05-14 by sequencer