chipsalliance/firrtl #1678: CircuitForm is deprecated by CircuitState isn't?

ekiwi-sifive, 2020-06-10:
It seems like `CircuitForm` is deprecated now, but somehow `CircuitState` is not.
This seems like an oversight, since `CircuitState` has a `CircuitForm` field.
@seldridge should `CircuitState` be deprecated?

seldridge, 2020-06-10:
We should eventually remove the `form: CircuitForm` member from `CircuitState`. This may have to be a 1.5 change to go through a full deprecation cycle.

Immediately, the apply methods for the `CircuitState` object that include `forms: CircuitForm` can be deprecated. Because I deprecated `CircuitForm`, I added an apply method that doesn't use it (so that we can get around deprecation warnings for the time being).

Fully removing `CircuitState` would mean that the underlying datatype for transforms would have to change. While I think this is fine to consider, I'm not sure it's worth it.

ekiwi-sifive, 2020-06-30:
>  I added an apply method that doesn't use it

I think this is what I was searching for when I created this issue.

closed 2020-06-30 by ekiwi-sifive