chipsalliance/firrtl #1687: firrtl.graph.PathNotFoundException: Unreachable node

IAMAl, 2020-06-12:
### Checklist

- current behavior?
nfo] [0.002] Elaborating design...
[info] [22.838] Done elaborating.
[error] (run-main-0) firrtl.FirrtlInternalException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error] firrtl.FirrtlInternalException: Internal Error! Please file an issue at https://github.com/ucb-bar/firrtl/issues
[error]         at firrtl.Utils$.error(Utils.scala:445)
[error]         at firrtl.Utils$.throwInternalError(Utils.scala:164)
[error]         at firrtl.stage.FirrtlStage.run(FirrtlStage.scala:35)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.Stage$$anon$1.transform(Stage.scala:43)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:140)
[error]         at firrtl.options.Translator.transform$(Phase.scala:140)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:46)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:46)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:168)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error]         at logger.Logger$.makeScope(Logger.scala:166)
[error]         at firrtl.options.Stage.transform(Stage.scala:46)
[error]         at chisel3.stage.phases.MaybeFirrtlStage.$anonfun$transform$2(MaybeFirrtlStage.scala:19)
[error]         at scala.Option.getOrElse(Option.scala:189)
[error]         at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:19)
[error]         at chisel3.stage.phases.MaybeFirrtlStage.transform(MaybeFirrtlStage.scala:13)
[error]         at chisel3.Driver$.$anonfun$execute$1(Driver.scala:226)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at chisel3.Driver$.execute(Driver.scala:226)
[error]         at chisel3.Driver$.execute(Driver.scala:249)
[error]         at en.ENMain$.delayedEndpoint$en$ENMain$1(ElectronNest.scala:339)
[error]         at en.ENMain$delayedInit$body.apply(ElectronNest.scala:338)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.App.main(App.scala:80)
[error]         at scala.App.main$(App.scala:78)
[error]         at en.ENMain$.main(ElectronNest.scala:338)
[error]         at en.ENMain.main(ElectronNest.scala)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.base/java.lang.reflect.Method.invoke(Method.java:566)
[error] Caused by: firrtl.graph.PathNotFoundException: Unreachable node
[error]         at firrtl.graph.DiGraph.path(DiGraph.scala:221)
[error]         at firrtl.graph.DiGraph.path(DiGraph.scala:203)
[error]         at firrtl.transforms.CheckCombLoops.$anonfun$expandInstancePaths$2(CheckCombLoops.scala:171)
[error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)

- [ ] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### Environment

- Chisel Verions: 
- OS:

sequencer, 2020-06-12:
Hi,would you mind providing chisel/firrtl code to reproduce this?

IAMAl, 2020-06-12:
Hi,
Difficult to reproduce with simple code, (I do not know where makes the exception).
But sharing my private repo is OK.

sequencer, 2020-06-12:
Sure in you don’t mind, you can invite me for debugging.

IAMAl, 2020-06-13:
Hi, I did.

IAMAl, 2020-06-13:
sbt 'runMain en.ENMain' produces the exception.

nrother, 2020-10-05:
Same issue here, with a relative simple design around a BlackBox. Are there any updates on this? Otherwise I could try to extract a minimal example from my code.

nrother, 2020-10-05:
The issue seems to be related to directly connecting a BlackBox output to an input of the same BlackBox (through one additional layer of hierachie in my case)

closed 2020-10-12 by IAMAl