chipsalliance/firrtl #1696: Add Protobuf FIRRTL Emitters

seldridge, 2020-06-16:
Add a new API that enables users to include FIRRTL emitters that write protobuf.

### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Add the ability for users to include protobuf emitters like how they include FIRRTL emitters.

As one API idea, a user could do the following to write low FIRRTL to a `Foo.pb` file.

```bash
firrtl -i Foo.fir -X low -o Foo.pb
```

### Type of Feature

  - new feature/API                    

### Related Features

A user can manually do this with `firrtl.proto.ToProto.writeToStream`. However, no protobuf emitter transform exists making protobuf emission a second-class citizen.

### External Information

This was requested on Gitter/Matrix ([permalink](https://view.matrix.org/room/!FQBlZBRjRMgQBRocZx:matrix.org/?anchor=$issOVoHTC0pkcj61bwCOsUW1OUpEadOqeSsdQhoUSRo&offset=-10&highlight=$issOVoHTC0pkcj61bwCOsUW1OUpEadOqeSsdQhoUSRo)).

jackkoenig, 2020-06-16:
One issue with deriving it from the file extension of the output file is it doesn't work with `-e`, so I think it needs to be a separate argument like `--emit-proto|-pb` or something.

seldridge, 2020-06-16:
Yeah, good point.

The unambiguous option here is via `--custom-transforms` and with an output file annotation that's sensitive to a specific emitter. Since we now support multiple emitters, a single output file really isn't expressive enough for supporting multiple emitters.

I do think a simpler API could be supported like: `-Xlow:proto:modules:Foo.bar` or `-Xhigh:firrtl:circuit:Baz.qux`. These are equivalent to a more verbose option format, e.g., `--custom-transforms LowFirrtlEmitter --emitter-option LowFirrtlEmitter:proto --emitter-option LowFirrtlEmitter:modules --emitter-option LowFirrtlEmitter:filenameBase:Foo.bar`. But, there's a clear expansion of simpler, shorthand options to more complex ones.

PR #2271 (2021-06-17): Add Protocol Buffer emission 
This adds infrastructure and annotations that let a user emit a FIRRTL
circuit as a Protocol Buffer.

Fixes #1696.

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [n/a] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a desired merge [... cut]
commit d708d3f0 (2021-06-17): Add Protocol Buffer emission  (#2271)

* Add Protocol Buffer emission export

This adds infrastructure and annotations that let a user emit a FIRRTL
circuit as a Protocol Buffer.

Fixes #1696.

* fixup! Add Protocol Buffer emission export
commit 5faa57c4 (2021-06-17): Add Protocol Buffer emission  (#2271)

* Add Protocol Buffer emission export

This adds infrastructure and annotations that let a user emit a FIRRTL
circuit as a Protocol Buffer.

Fixes #1696.

(cherry picked from commit d708d3f0555bb940ff417d3574c6902bb7a9c853)
commit 56f1de86 (2021-06-17): Add Protocol Buffer emission  (#2271)

* Add Protocol Buffer emission export

This adds infrastructure and annotations that let a user emit a FIRRTL
circuit as a Protocol Buffer.

Fixes #1696.

(cherry picked from commit d708d3f0555bb940ff417d3574c6902bb7a9c853)
closed 2021-06-17 by mergify[bot]