chipsalliance/firrtl #1702: Could I get LowFirrtl from HighFirrtl form?

zhaoxueyan1, 2020-06-21:
I wanted to get LowFirrtl from the souce ./firrtl/gcd.fir, but i got a bug in Utils.scala with **"Internal Error! flow: shouldn't be here - Reference(x,UIntType(IntWidth(32)))"**.
The code is as follows:

```scala
val circuit = firrtl.Parser.parse(Source.fromFile(inputFile).getLines,
      firrtl.Parser.IgnoreInfo)
    val  topName = circuit.main
    var state = CircuitState(circuit, firrtl.ChirrtlForm)
    val ChiToHigh = new ChirrtlToHighFirrtl()
    val HighToMid = new HighFirrtlToMiddleFirrtl()
    val MidToLow = new MiddleFirrtlToLowFirrtl()
    val analyzeCircuit = new AnalyzeCircuit()
//
    state = ChiToHigh.execute(state)
    state = HighToMid.execute(state)
    state = MidToLow.execute(state)
    analyzeCircuit.execute(state)
``` 
The bug happened in the following function:
```scala
def flow(e: Expression): Flow = e match {
    case ex: WRef => ex.flow
    case ex: WSubField => ex.flow
    case ex: WSubIndex => ex.flow
    case ex: WSubAccess => ex.flow
    case ex: DoPrim => SourceFlow
    case ex: UIntLiteral => SourceFlow
    case ex: SIntLiteral => SourceFlow
    case ex: Mux => SourceFlow
    case ex: ValidIf => SourceFlow
    case WInvalid => SourceFlow
    case ex => throwInternalError(s"flow: shouldn't be here - $e")
  }
```

zhaoxueyan1, 2020-06-21:
Gcd.fir
```Verilog
circuit GCD :
  module GCD :
    input clock : Clock
    input reset : UInt<1>
    output io : {flip a : UInt<32>, flip b : UInt<32>, flip e : UInt<1>, z : UInt<32>, v : UInt<1>}
    reg x : UInt<32>, clock
    reg y : UInt<32>, clock
    node _T_14 = gt(x, y)
    when _T_14 :
      node _T_15 = sub(x, y)
      node _T_16 = tail(_T_15, 1  )
      x <= _T_16
    node _T_18 = eq(_T_14, UInt<1>("h00"))
    when _T_18 :
      node _T_19 = sub(y, x)
      node _T_20 = tail(_T_19, 1)
      y <= _T_20
    when io.e :
      x <= io.a
      y <= io.b
    io.z <= x
    node _T_22 = eq(y, UInt<1>("h00"))
    io.v <= _T_22

```

seldridge, 2020-06-21:
You need to convert the IR to working IR. However, it would be better if you used the new infrastructure to build a compiler with your specific target (low form) and current state (Chirrtl):

```scala
val compiler = new firrtl.stage.transforms.Compiler(
  targets=firrtl.stage.Forms.LowForm,
  currentState=firrtl.stage.Forms.ChirrtlForm
)
val statex = compiler.transform(state)

zhaoxueyan1, 2020-06-22:
With @seldridge 's help, it has worked now. Thank you very much.

seldridge, 2020-06-22:
Great!

For a little more info, your approach should also work, but would need to run [`IRToWorkingIR`](https://github.com/freechipsproject/firrtl/blob/9ff347c48eef530be9cbf1f8e5bbfb9ed053d182/src/main/scala/firrtl/LoweringCompilers.scala#L32) and [`ResolveAndCheck`](https://github.com/freechipsproject/firrtl/blob/9ff347c48eef530be9cbf1f8e5bbfb9ed053d182/src/main/scala/firrtl/LoweringCompilers.scala#L45) between `ChirrtlToHighFirrtl` and `HighFirrtlToMiddleFirrtl`. Since those are deprecated and slated to get removed, my original suggestion is the advised approach.

I'll close this, but feel free to reopen if there's some lingering issue here.

closed 2020-06-22 by seldridge