chipsalliance/firrtl #1744: TopWiringTransform Doesn't Respect Direction

seldridge, 2020-07-09:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [not really] Did you provide a code example showing the problem?
- [not really] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

If you try to top wire a flipped wire you get an output port. 

### What is the expected behavior?

Top wiring should respect directionality.

### Steps to Reproduce

(Tracking issue for my own benefit. I'll populate this if I un-assign myself.)

### Your environment

1.3.x

### External Information

None.

davidbiancolin, 2020-07-10:
Unfortunately, i think the destruction of Flipped-ness is what Alon wanted though as in FireSim we use this exclusively to coerce a bunch of signals (that were generally tapping for instrumentation) to output. Obviously, the transform needs to have schedule-independent behavior, and i think your expectation is that it respect direction is the natural one, but can we support selecting from the two modes?

seldridge, 2020-07-10:
Optionally choosing one or the other at the FIRRTL transform level is possible, but it's hacky:

1) (Like in the example code above...) You can add a transform that depends on `LowerTypes` with an `optionalPrerequisiteOf` pointing at `TopWiringTransform`.
2) Use inheritance, or a wrapper with an underlying `TopWiringTransform`, that uses different dependencies, either add `LowerTypes` as a prerequisites for direction destroying or the contents of #1750 for direction preserving.

That said... I think this be better handled at the Chisel level with one of:

1) Explicitly wrapping `wire` in `Output` to destroy flippedness if you want signal watching behavior
2) Use a top wiring API that has methods for one or the other, e.g., `def drill(a: Data)` to top wire with directions and `def watch(a: Data)` to top wire directionless.

Any thoughts on these workarounds for FireSim? Any input @alonamid?

davidbiancolin, 2020-07-11:
Chisel-side API changes aren't going to work for us as we run top wiring at multiple points during compilation, during which we're converting some other annotation into a TopWiringAnnotation and invoking the transformation. Seems to me adding LowerTypes as a prerequisite of these aforementioned enclosing transforms (per (2)) is a sufficient workaround. 

Is #1750 back-portable?

seldridge, 2020-07-11:
> Is #1750 back-portable?

#1750 is backportable to 1.3.x and seems to make sense to do so as it fixes the ordering bug (or defines a scheduling-independent behavior).

A user could hit the bug on 1.2.x but that's more of explicit user error and not really fixable. I haven't tested this, but I think if a user had an  `outputForm=LowForm` custom transform (that was internally running `LowerTypes`) then you would get the directionless behavior with `-fct custom,TopWiringTransform` and direction-preserving behavior with `-fct TopWiringTransform,custom`. 

> ... during which we're converting some other annotation into a TopWiringAnnotation and invoking the transformation

Hmm... if FireSim is invoking `TopWiringTransform` directly and not relying on FIRRTL to schedule it, then maybe #1750 wouldn't make a difference one way or another. It sounds like FireSim isn't using either `-fct TopWiringTransform` or a `RunFirrtlTransformAnnotation` for scheduling. Is that right?

Regardless, if you need something immediate to use, the following will force the scheduling behavior one way or another:

```scala
/** Modified TopWiringTransform that ignores flips when top-wiring */
class TopWiringDirectionless extends TopWiringTransform {
  override def prerequisites = Dependency(LowerTypes) +: super.prerequisites
}

/** Modified TopWiringTransform that preserves flips when top-wiring. */
TopWiringPreserveDirections extends Transform with DependencyAPIMigration {

    private val underlying = new TopWiringTransform

    override def prerequisites = underlying.prerequisites
    override def optionalPrerequisites = underlying.prerequisites
    override def optionalPrerequisiteOf = Dependency(LowerTypes) +: underlying.optionalPrerequisiteOf
    override def invalidates(a: Transform) = underlying.invalidates(a)

    override def execute(a: CircuitState) = underlying.execute(a)

}
```

Note: the type of `optionalPrerequisiteOf`, normally `Seq[Dependency[Transform]]`, was accidentally specialized to `Seq[Dependency[Emitter]]` and this is preventing overriding.

davidbiancolin, 2020-07-11:
> > Is #1750 back-portable?
> 
> #1750 is backportable to 1.3.x and seems to make sense to do so as it fixes the ordering bug (or defines a scheduling-independent behavior).
 
Indeed. 

> Hmm... if FireSim is invoking `TopWiringTransform` directly and not relying on FIRRTL to schedule it, then maybe #1750 wouldn't make a difference one way or another. It sounds like FireSim isn't [...]

 There's one case where we're dynamically adding a transform that invokes TopWiringTransform (using our own mechanism, not the two you gave) -- so long as that transform defines the right prerequisites there will be no surprises. All of the other instances of this are part of the core compiler and are statically scheduled in a mega SeqTransform like thing -- those should will likely be fine. 

> Regardless, if you need something immediate to use, the following will force the scheduling behavior one way or another:
>
> ```scala
> /** Modified TopWiringTransform that ignores flips when top-wiring [...]

Thanks for the tips!

alonamid, 2020-07-11:
I'm curious, what is your use-case for an input-direction TopWiring transform that shouldn't be handled suing the Wiring Transform?

seldridge, 2020-07-11:
It's to expose a port of a submodule and drive it from the outside. Roughly:

```scala
val foo = Module(new Foo { val port = IO(new Bar) } )
val baz = Wire(chiselTypeOf(subModule.port))
topWire(baz)
```

`TopWiringTransform` was convenient because it could be done from the point where the module was instantiated as opposed to having to know the type of the module port and do sources/sinks for the `WiringTransform`.

This worked at some point, then after accumulating some custom transforms, I noticed that directions started getting stripped.

PR #1750 (2020-07-14): Make TopWiringTransform run before LowerTypes
Add an optionalPrerequisiteOf to TopWiringTransform pointing at
LowerTypes.

This fixes a bug where top-wired wire bundles with flips could result
in the top-wired, flattened bundle having every field with an "output"
direction if the TopWiringTransform is moved around in the transform
order (see FIRRTL issue #1744).

Why did this happen?

Fundamentally, this stems from the fact that LowerTypes preserves
bundle direction for ports, but destroys it for wires.

Specifically, The TopWiringTransform creates ports of the "output"
direction that are copi [... cut]
closed 2020-07-14 by mergify[bot]