chipsalliance/firrtl #1935: Generate some code only for System Verilog backend

Martoni, 2020-10-28:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

I'm using the trait Formal in [my module](https://github.com/Martoni/GbVga/blob/master/chisel/src/main/scala/gbvga/gbwrite.scala#L12) description. It come from [chisel-formal](https://github.com/tdb-alcorn/chisel-formal) project.
Formal specific code (assert(), assume(), ...)  is SystemVerilog language and to generate it I'm using the argument `-X sverilog`.
It works for System Verilog file `*.sv` but specific code is also generated for normal verilog file `*.v`.
I have to comment the formal code to avoid it in simulation (because it generate errors).

Is it possible to add an option to not generate it in standard Verilog ?

### Type of Feature

 - backend code generation
 - new feature/API

### Related Features

Related to chisel module chisel-formal.

### External Information

Issue added in chisel-formal github too. https://github.com/tdb-alcorn/chisel-formal/issues/12

ekiwi, 2020-10-30:
Could you please attach the Verilog file that caused verilator to throw an error?

Martoni, 2020-10-30:
For my project [GbVga](https://github.com/Martoni/GbVga), I'm using chisel-formal module with trait «with Formal» : 
```Scala
class GbWrite (val datawidth: Int = 2,
               val debug_simu: Boolean = true,
               val aformal: Boolean = false) extends Module with Formal {
//...
}
```
This code display a warning : 
```
------------------------------------------------------------------------------
Warning: 5 verification statements (assert, assume or cover) were removed when compiling to Verilog because the basic Verilog standard does not support them. If this was not intended, compile to System Verilog instead using the `-X sverilog` compiler flag.
------------------------------------------------------------------------------
```
 when I generate standard verilog with chiselstage : 
```Scala
  println("> generate verilog")
  (new ChiselStage).execute(args,
      Seq(ChiselGeneratorAnnotation(() => new GbWrite(8, debug_simu=true))))
```

And that adding this code instantiation : 
```Verilog
  ResetCounter ResetCounter ( // @[Formal.scala 14:36]
    .clock(ResetCounter_clock),
    .reset(ResetCounter_reset),
    .numResets(ResetCounter_numResets),
    .timeSinceReset(ResetCounter_timeSinceReset)
  );
  TestCase TestCase ( // @[Formal.scala 20:32]
    .testCase(TestCase_testCase)
  );
```
Which generate verilator compilator error in my [CocoTB make process](https://github.com/Martoni/GbVga/tree/master/cocotb/test_gbwrite) : 
```Shell
_gbwrite/GbWrite_withtp.v /home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/cocotb/share/lib/verilator/verilator.cpp
%Error-TIMESCALEMOD: /media/stockage/projets/GbVga/cocotb/test_gbwrite/GbWrite_withtp.v:47:16: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /media/stockage/projets/GbVga/cocotb/test_gbwrite/GbWrite_withtp.v:3:8: ... Location of module with timescale
    3 | module GbWrite(
      |        ^~~~~~~
%Error-TIMESCALEMOD: /media/stockage/projets/GbVga/cocotb/test_gbwrite/GbWrite_withtp.v:53:12: Timescale missing on this module as other modules have it (IEEE 1800-2017 3.14.2.2)
                     /media/stockage/projets/GbVga/cocotb/test_gbwrite/GbWrite_withtp.v:3:8: ... Location of module with timescale
    3 | module GbWrite(
      |        ^~~~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual and https://verilator.org for more assistance.
make[1]: *** [/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:42: sim_build/Vtop.mk] Error 1
make[1]: Leaving directory '/media/stockage/projets/GbVga/cocotb/test_gbwrite'
make: *** [/home/fabien/pyenv/pyenv37/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
```

It's a pity because this «pure verilog» source is not used for Formal verification as explain in the warning. To do formal verification I'm generating system verilog file *.v with chisel stage following code:
```Scala
  println("> generate systemVerilog for formal")
  (new ChiselStage).execute(Array("-X", "sverilog"),
      Seq(ChiselGeneratorAnnotation(() => new GbWrite(8, aformal=true))))
```

ekiwi, 2020-10-30:
> It's a pity because this «pure verilog» source is not used for Formal verification as explain in the warning.

It seems to me like the two errors that verilator is showing are unrelated to the formal statements. They are both complaining about missing timescales on some modules, while apparently another module has them. This should be unrelated to firrtl, since firrtl does never add any timescales. Could it be that cocotb is trying to add a timescale, but only does so for the first module?

Martoni, 2020-10-31:
>It seems to me like the two errors that verilator is showing are unrelated to the formal statements.

If I comment the code «with Formal» the two instantiations of «  ResetCounter» and «TestCase» disappear.

> They are both complaining about missing timescales on some modules, while apparently another module has them. This should be unrelated to firrtl, since firrtl does never add any timescales.

They complaining about missing timescale under «ResetCounter» and «TestCase» module that are useless in the Verilog file, because they are used for Formal verification only.

> Could it be that cocotb is trying to add a timescale, but only does so for the first module?

Exactly (in fact it's not cocotb but me in the [makefile](https://github.com/Martoni/GbVga/blob/master/cocotb/test_gbwrite/Makefile#L26)) 

That why I need a verilog generated file without these instantiation. They are useless for «normal» simulation and for synthesis. These instantiations should be generated only in system verilog file (-X sverilog).

ekiwi, 2020-10-31:
> That why I need a verilog generated file without these instantiation. They are useless for «normal» simulation and for synthesis. These instantiations should be generated only in system verilog file (-X sverilog).

Oh I see. Then this isn't a firrtl bug, as I initially thought, but actually a missing feature in `chisel-formal`. The only thing that `firrtl` removes when you compile in Verilog mode are the `assert`, `assume` and `cover` statements. It won't remove the modules that were added by `chisel-formal`.

closed 2020-10-31 by ekiwi