chipsalliance/firrtl #1959: Bug in RemoveAccess?

sequencer, 2020-11-14:
Seems to be a bug that `RemoveAccess` failed to remove this corner case. 
### Checklist
- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?
```
java.lang.RuntimeException: LowerTypes expects all SubAccesses to have been expanded! bankHitWays[bankIdxInOrder[0]]
```

### What is the expected behavior?
emit verilog successfully.
 
### Steps to Reproduce
https://scastie.scala-lang.org/a6R4pPCMRreFhFYzWcXMXw

### Your environment
Chisel 3.4.0
FIRRTL 1.4.0
OpenJDK 14
### External Information

ekiwi, 2020-11-16:
Sound to me like there is a problem in the `RemoveAccesses` pass.

jackkoenig, 2020-12-02:
Fixed by https://github.com/chipsalliance/firrtl/pull/1984

closed 2020-12-02 by jackkoenig