chipsalliance/firrtl #1992: Fileinfo specification

dalance, 2020-11-30:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

In spec.pdf, fileinfo is defined as `info = @[string]`, but the format of `string` is not defined.
I think this missing will cause compatibility issues with other front-ends except Chisel in future.
Additionally, the current string format of Chisel lacks absolute file path.
This means back-ends of FIRRTL can't find the original source codes.

So I think the specification of fileinfo should be defined like below:

```
info = @[ [[absolute_file_path]] [[line]]:[[row]] ]
absolute_file_path = string
line = int
row = int
```

This specification is an idea, and I think it can be improved.
For example, file path encoded as URI. It is described at https://github.com/chipsalliance/firrtl/issues/1695#issuecomment-645501722.

The debug information of LLVM-IR may be helpful.

### Type of Feature

- documentation

### Related Features

https://github.com/chipsalliance/firrtl/issues/1290

### External Information

https://llvm.org/docs/SourceLevelDebugging.html#c-c-front-end-specific-debug-information

ekiwi, 2020-12-02:
Currently the format is mostly aimed at direct human consumption.

Imho if we want to include the full path info, we cannot just include it at the end of every line, as that would more than double
the size of firrtl files. Instead we would probably want to have a global lookup table similar to LLVM. This will not be an obstacle to tools and might even reduce the size of the firrtl file (since every file name would only have to be written out exactly once). However, it will make it much harder for humans to benefit from the information, since they will need to resolve the indirection manually.

chick, 2020-12-03:
Rendering the putative machine friendly source information would be a good addition to the IntelliJ firrtl plugin