chipsalliance/firrtl #2012: The end of the LoFIRRTL file should end with DEDENT

itviewer, 2020-12-08:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
According to the FIRRTL.g4 Grammar
```
circuit
  : 'circuit' id ':' info? INDENT module* DEDENT EOF
  ;

```
The `.fir ` file should end with DEDENT EOF, but the `.lo.fir` file does not end with DEDENT.

### What is the expected behavior?
just like `.fir ` file, the `.lo.fir` file should end with DEDENT

### Steps to Reproduce

Any Chisel circuit module to generate .fir and .lo.fir files

### Your environment

- Chisel Verions: 3.4.+
- Firrlt Version: 1.4.+

### External Information

Firrtl [Intellij plugin ](https://plugins.jetbrains.com/plugin/14183-easysoc-firrtl)  will check whether the `.fir` file conforms to the specification according to FIRRTL.g4 Grammar.
[image]

sequencer, 2020-12-08:
quite interesting, I notice that `lo.fir` also can be parsed with antlr4, wondering why FIRRTL can parse it while your plugin cannot. I'll investigate.

itviewer, 2020-12-08:
> quite interesting, I notice that `lo.fir` also can be parsed with antlr4, wondering why FIRRTL can parse it while your plugin cannot. I'll investigate. 

the plugin can parse both `.fir`and `.lo.fir`，the problem is the latter does not strictly satisfy the grammar.

itviewer, 2020-12-10:
When serializing the circuit state, [Serializer](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/ir/Serializer.scala) uses the `doIndent` method to deal with DEDENT.  `s(node: Circuit)` method lacks processing of the end of the file. We need to add a newline with `newLineNoIndent()` to satisfy the Grammar.
```
  private def s(node: Circuit)(implicit b: StringBuilder, indent: Int): Unit = node match {
    case Circuit(info, modules, main) =>
      b ++= "circuit "; b ++= main; b ++= " :"; s(info)
      if (modules.nonEmpty) {
        newLineAndIndent(1); s(modules.head)(b, indent + 1)
        modules.drop(1).foreach { m => newLineNoIndent(); newLineAndIndent(1); s(m)(b, indent + 1) }
      }
     newLineNoIndent()
  }
```

closed 2020-12-10 by mergify[bot]