chipsalliance/firrtl #2013: ResolveFlows Accepts Invalid FIRRTL IR

seldridge, 2020-12-08:
### Checklist

- [ ] Did you specify the current behavior?
- [ ] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?
`firrtl.passes.ResolveFlows` sets the flows of ports and instance ports incorrectly relative to the spec. This results in the Scala FIRRTL Compiler accepting circuits that allow using an output port or input instance port (sink flow) as a RHS connection (source flow).

While this has been allowed forever in Chisel, it should be disallowed in the Scala FIRRTL Compiler (SFC) and supported with a conversion transform at the CHIRRTL IR level.

The spec states:

For flow, the spec states (Section 8):

> A reference to an input port, an instance, a memory, and a
> node, is a source. A reference to an output port is a sink.

For a connect to be legal, the spec states (Section 5.1):

> Either the flow of the right-hand side expression is source or
> duplex, or the right-hand side expression has a passive type.

These two together should disallow using an output port as a RHS connection, but the SFC allows it:

```
# cat Foo.fir
circuit Foo:
  module Foo:
    output a: UInt<1>
    output b: UInt<1>
    
    a <= UInt<1>(0)
    b <= a
# cat Foo.v
module Foo(
  output  a,
  output  b
);
  assign a = 1'h0;
  assign b = a;
endmodule
```

For an instance, the base reference to the instance has source flow, and the orientations of ports are flipped. The following RHS usage of `baz.a` should be disallowed because it has sink flow:

```
# cat Bar.fir
circuit Bar:
  module Baz:
    input a: UInt<1>
  module Bar:
    output a: UInt<1>
    inst baz of Baz
    
    baz.a <= UInt<1>(0)
    a <= baz.a
# cat Bar.v
module Baz(
  input   a
);
endmodule
module Bar(
  output  a
);
  wire  baz_a;
  Baz baz (
    .a(baz_a)
  );
  assign a = baz_a;
  assign baz_a = 1'h0;
endmodule
```

### What is the expected behavior?

While this has always been a feature of Chisel, it should be disallowed inside the Scala FIRRTL Compiler. This can continue to be supported for Chisel, but it should be removed as part of CHIRRTL removal.

### Steps to Reproduce

Compile the above circuits expecting them to fail. They don't.

### Your environment

Scala FIRRTL Compiler 1.4.0.

### External Information

None.

ekiwi, 2020-12-08:
Why can't we change the spec? In C you can assign to a variable and also read its output, so why should that not be true for output ports? Why can't an output just be treated like a Wire that cannot be deleted?

seldridge, 2020-12-08:
Changing the spec is an option. However, I would argue that allowing this is bad IR design that's leaking Chisel semantics. Modules are closer to "functions" and C or LLVM IR isn't going to let you read a callee's input or your own return value after the return. We have great constructs for this already: nodes, wires, and registers. I'd rather have a simpler IR than special cased weirdness like:

> an output [is] treated like a Wire that cannot be deleted

Transforms being easier to write if the spec invariant holds (Jack has a custom transform that converts input reads to wires to make a later transform simpler). Also, this makes it way easier to write another implementation of a FIRRTL compiler. Connect validation can be done just by looking at the type as opposed to special casing based on the kind.

_No proposal to change this semantic in Chisel_, just to change the Scala FIRRTL Compiler to include a transform that removes this behavior as part of CHIRRTL -> High lowering.

ekiwi, 2020-12-08:
> Transforms being easier to write if the spec invariant holds (Jack has a custom transform that converts input reads to wires to make a later transform simpler). Also, this makes it way easier to [...]

To me it seems like treating outputs as wires actually removes a special case. When I was writing the SMT backend I was annoyed by the fact that the RemoveWires transform treats outputs differently and does not put assignments to them into SSA order.
If you look at the btor2 format, they just treat outputs as labeled signals which I believe is the simplest way to deal with them.

ekiwi, 2020-12-08:
> Modules are closer to "functions" and C or LLVM IR

I am not sure how modules are functions, since there is no call hierarchy and the evaluation order between modules is non-trivial. I.e. if you want to treat submodule evaluation as a function call, you will have to do a fixed point computation, potentially evaluating the submodule multiple times.

lattner, 2020-12-20:
FWIW, inout ports (also Analog types) seem semantically more close to wires than output ports.  There is a precedent for what output means in the Verilog that is worth matching.  It's true that at the end of they day, these are just wires that can be driven from both sides, but modeling it correctly in the IR seems valuable.