chipsalliance/firrtl #2029: div(a, a) mismatch

drom, 2020-12-27:
### Checklist

- [ ] Did you specify the current behavior?
- [ ] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?

The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_1: UInt<14>
    output tmp4: UInt<14>
    tmp4 <= div(inp_1, inp_1)
```

Compiled with `firrtl-1.4.0` generates the following Verilog

```verilog
module top_mod(
  input  [13:0] inp_1,
  output [13:0] tmp4
);
  assign tmp4 = 14'h1;
endmodule
```

### What is the expected behavior?

Expected the following Verilog

```verilog
module top_mod(
  input  [13:0] inp_1,
  output [13:0] tmp4
);
  assign tmp4 = inp_1 / inp_1;
endmodule
```

Formal tool `Yosys (Yosys 0.9+3755 (git sha1 442d19f6, clang 11.0.0 -fPIC -Os))` reports mismatch between current and expected result:

```
ERROR: Found 13 unproven $equiv cells in 'equiv_status -assert'.
```

### Steps to Reproduce

`./firrtl-1.4.0 -i top_mod.fir`

### Your environment

`Linux linux-sydi 5.9.14-1-default #1 SMP Sat Dec 12 06:57:32 UTC 2020 (c648a46) x86_64 x86_64 x86_64 GNU/Linux`

### External Information

https://twitter.com/wavedrom/status/1342932252531945472

https://github.com/llvm/circt/issues/363

aswaterman, 2020-12-28:
LEC fails between Firrtl versions sometimes, which is preferable to avoid but doesn't always indicate a bug.

In this case, I argue it's not a bug.  AFAICS, the only reason they are not equivalent is that `0/0` is `'x` in Verilog.  In HW, that is going to produce an arbitrary non-`'x` value, so returning 1 is a correct outcome.

In other words, if LEC fails here, then LEC would also fail for a gate-level netlist, because the netlist would obviously not produce `'x` in this case, either.

seldridge, 2020-12-28:
(I commented on the llvm/circt issue.) Note: that the expected verilog is produced with optimizations turned off:

```bash
firrtl -X mverilog -i top_mod.fir && cat top_mod.v
```

Produces:
```verilog
module top_mod(
  input  [13:0] inp_1,
  output [13:0] tmp4
);
  assign tmp4 = inp_1 / inp_1;
endmodule
```

jackkoenig, 2021-01-04:
Looking at the CIRCT issue, it looks like we're all on the same page that this is a desirable optimization despite it being a LEC failure. It would be nice if we had a way of expressing LEC-failing optimizations in a way that we could waive such differences but still check the rest of the design.

drom, 2021-01-04:
Is it desirable, because it is useful?

drom, 2021-01-04:
Could we agree on this?
```
assign tmp7 = (inp_0 == 0) ? 8'hff : 8'h1;
```
it is equivalent from Yosys LEC's point of view.

jackkoenig, 2021-01-04:
> Is it desirable, because it is useful?

Yes

> Do we have a lot of cases like `A/A`?

Probably not; something doesn't have to be common to be useful.

> If it is so useful, why the following example is not optimized?

Looks like we don't propagate direct renames across ports (only across wires and nodes). That's a separate issue that will block any `op(a, a)` optimization, not just this case.

> Could we agree on this?

Are you saying that according to Yosys:
```verilog
assign tmp7 = (inp_0 == 7'b0) ? 8'hff : 8'h1;
```
is equivalent to
```verilog
assign tmp7 = inp_0 / inp_0;
```
?

drom, 2021-01-04:
> Are you saying that according to Yosys:
> 
> ```verilog
> assign tmp7 = (inp_0 == 7'b0) ? 8'hff : 8'h1;
> ```
> 
> is equivalent to
> 
> ```verilog
> assign tmp7 = inp_0 / inp_0;
> ```
> 
> ?

Yes

ekiwi, 2021-01-04:
> Yes

Seems like that might only be an artifact of the SMTLib interpretation: https://github.com/chipsalliance/firrtl/blob/15013df6f6ac2dafeb35d7ed15cf95c7ac8a5bef/src/test/scala/firrtl/backends/experimental/smt/end2end/UndefinedFirrtlSpec.scala#L9

aswaterman, 2021-01-04:
It would be better _not_ to optimize than to use `? 8'hff : 8'h1`.  At least synthesis tools can optimize `a / a`, but they obviously can't optimize the other thing.

(IMO, we should continue optimizing `a / a` into 1...)

drom, 2021-01-05:
> It would be better _not_ to optimize than to use `? 8'hff : 8'h1`. At least synthesis tools can optimize `a / a`, but they obviously can't optimize the other thing.
> 
> (IMO, we should continue optimizing `a / a` into 1...)

I spend a day searching for the LEC mode that would agree with that. No success.
This is a big boulder preventing us from using Verilog LEC tools for comparing FIRRTL and MLIR/CIRCT results

jackkoenig, 2021-01-05:
Repeating what I said earlier: "It would be nice if we had a way of expressing LEC-failing optimizations in a way that we could waive such differences but still check the rest of the design."

Is there a way in Yosys to tell it to basically ignore this difference? While we certainly could work around this specific issue, it is not the only one and it feels a bit weird to have to either not do certain optimizations or do them in an arbitrary way just to make one specific tool's LEC happy.

Perhaps the simplest solution would be to provide an option to disable LEC-violating optimizations.

drom, 2021-01-05:
> Perhaps the simplest solution would be to provide an option to disable LEC-violating optimizations.

After running my FIRRTL fuzzer, I have found only one optimization that breaks Yosys LEC beyond repair, so far.
It would help me a lot having an option to disable `A / A -> 1`.

PR #2040 (2021-01-20): Add --dont-fold option to disable folding specific prim ops
This adds a --dont-fold options (backed by a DisableFold annotation)
that lets a user specify primitive operations which should never be
folded. This feature lets a user disable certain folds which may be
allowable in FIRRTL (or by any sane synthesis tool), but due to inane
Verilog language design causes formal equivalence tools to fail due to
the fold.

Add a test that a user can disable `a / a -> 1` with a
DisableFold(PrimOps.Div) annotation.

Fixes #2029.

### Example Usage

The following circuit:
```
circuit Foo :
  module Foo :
  [... cut]
closed 2021-01-20 by mergify[bot]