chipsalliance/firrtl #2030: Issue with Analog in 1.4.1

mwachs5, 2021-01-04:
### Checklist

- [x] Did you specify the current behavior?
- [ ] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?

I am seeing this exception on a design with Analog wires that I cannot post here. Will work on a more reduced test case:

```
Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! Unexpected type AnalogType(IntWidth(16))                                                                                                                                                                                                          Please file an issue at https://github.com/ucb-bar/firrtl/issues                                                                                                                                                                                                                                                                 at firrtl.Utils$.error(Utils.scala:472)                                                                                                                                                                                                                                                                                      at firrtl.Utils$.throwInternalError(Utils.scala:163)                                                                                                                                                                                                                                                                         at firrtl.Utils$.getGroundZero(Utils.scala:268)                                                                                                                                                                                                                                                                              at firrtl.transforms.RemoveWires.onStmt$1(RemoveWires.scala:133)                                                                                                                                                                                                                                                             at firrtl.transforms.RemoveWires.$anonfun$onModule$2(RemoveWires.scala:139)                                                                                                                                                                                                                                                  at firrtl.transforms.RemoveWires.$anonfun$onModule$2$adapted(RemoveWires.scala:139)                                                                                                                                                                                                                                          at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)                                                                                                                                                                                                                                                  at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)                                                                                                                                                                                                                                                 at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)                                                                                                                                                                                                                                                        at firrtl.ir.Block.foreachStmt(IR.scala:539)                                                                                                                                                                                                                                                                                 at firrtl.traversals.Foreachers$StmtForMagnet$$anon$1.foreach(Foreachers.scala:18)                                                                                                                                                                                                                                           at firrtl.traversals.Foreachers$StmtForeach$.foreach$extension(Foreachers.scala:35)
```

### What is the expected behavior?

This should not give an exception. This worked in 1.4.0 and started failing in 3.4.1

### Steps to Reproduce

Will work on a smaller test case

### Your environment

- Chisel Verions: 3.4.1., 1.4.1

### External Information

mwachs5, 2021-01-04:
Well the error message matches the scaladoc anyway:

```
  /** Returns an [[firrtl.ir.Expression Expression]] equal to zero for a given [[firrtl.ir.GroundType GroundType]]
    * @note Does not support [[firrtl.ir.AnalogType AnalogType]] nor [[firrtl.ir.IntervalType IntervalType]]
    */
  def getGroundZero(tpe: GroundType): Expression = tpe match {
```

mwachs5, 2021-01-05:
Note that for me a workaround is to just make the val a Analog(16.W) instead of a Wire(Analog(16.W)) (I needed it to have the right type, but getting rid of the Wire(...) made it work. Assigning it to a `DontCare` did not help.

jackkoenig, 2021-01-05:
Mea culpa, this was almost certainly introduced in https://github.com/chipsalliance/firrtl/pull/1949