chipsalliance/firrtl #2052: (div, cat) mismatch

drom, 2021-01-21:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?

The following FIRRTL program.

```python
; seed: 371
circuit top_mod :
  module top_mod :
    input tmp8: UInt<13>
    input tmp11: UInt<11>
    output tmp40: UInt<24>
    wire tmp28: UInt<13>
    tmp28 <= div(tmp8, UInt<20>(1))
    tmp40 <= cat(tmp11, tmp28)
```

Running with `firrtl-1.5.0-` generates the following Verilog

```verilog
module top_mod(
  input  [12:0] tmp8,
  input  [10:0] tmp11,
  output [23:0] tmp40
);
  wire [19:0] _GEN_0 = {{7'd0}, tmp8};
  wire [19:0] tmp28 = _GEN_0 / 20'h1;
  assign tmp40 = {tmp11,tmp28};
endmodule
```
### What is the expected behavior?

```verilog
module top_mod(
  input  [12:0] tmp8,
  input  [10:0] tmp11,
  output [23:0] tmp40);

  wire [12:0] tmp28;	// a_top_mod.fir:7:5

  assign tmp28 = tmp8;	// a_top_mod.fir:8:11
  assign tmp40 = {tmp11, tmp28};	// a_top_mod.fir:3:3, :9:14
endmodule
```

Which is not formally equivalent

```
ERROR: Found 11 unproven $equiv cells in 'equiv_status -assert'.
```

### Steps to Reproduce

`./firrtl-1.5.0 -i top_mod.fir`

### Your environment

`Linux linux-sydi 5.9.14-1-default #1 SMP Sat Dec 12 06:57:32 UTC 2020 (c648a46) x86_64 x86_64 x86_64 GNU/Linux`

- Verilator version: `Verilator 4.103 devel rev v4.102-42-g726e78fd`

### External Information

aswaterman, 2021-01-21:
I agree the "expected behavior" looks better, but these really do look equivalent.  Is there a counterexample?

aswaterman, 2021-01-21:
The width of the `div` should be truncated to the smaller of the widths, so the "expected" behavior looks correct."

drom, 2021-01-21:
yes, `tmp28` becoming a 20-bit number screwing up the concatenation.

PR #2058 (2021-01-29): Stop padding multiply and divide ops
Fixes #2052
Also fixes some Verilog lint issues

A lot of work for a change that is literally deleting what 12 characters?

I'm a bit paranoid about this change so I wrote some formal checks, but notably I'm not checking every width, and I'm relying on the reference Verilog being correct. Any thoughts?

As this is a correctness issue, at least the fix should go all the way back.

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [NA] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you ad [... cut]
closed 2021-01-29 by jackkoenig