chipsalliance/firrtl #2078: Port Names With Verilog Keywords Are Not Renamed in Verilog Module Instantiations

davidbiancolin, 2021-02-09:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
Instance port names that are Verilog keywords are not properly renamed but Module port names are. 

### What is the expected behavior?
Instance port names should match the module port names after verilog emission. 

### Steps to Reproduce
Lower these modules to veriog:

```
class Child extends MultiIOModule {
  val time = IO(Output(Bool()))
  dontTouch(time)
  time := false.B
}

class Parent extends MultiIOModule {
  val time = IO(Output(Bool()))
  val child = Module(new Child)
  time := child.time
}
```

### Your environment
See this scastie environment: https://scastie.scala-lang.org/174jJgvJQwWQL0cAn07otA

- Chisel Version: 3.4.2

### External Information

The example above produces the following illegal verilog:
```
module Child(
  output  time_,
  input   casez_
);
  assign time_ = casez_;
endmodule
module Parent(
  output  time_,
  input   casez_
);
  wire  child_time;
  wire  child_casez;
  Child child (
    .time(child_time),
    .casez(child_casez)
  );
  assign time_ = child_time_;
  assign child_casez_ = casez_;
endmodule
```

This breakage was introduced with the rewrite of RemoveKeywordCollisions (33bdea461c22d9941b061bab3d4a86f4b16d01f7) and affects Versions 1.4.*.

I initially thought this was a `ManipulateNames` bug but it seems like what's actually happening is that Port renames in `ManipulateNames` aren't being fully respected. Port definitions in `Module`s are being successfully renamed. However, the `tpe` fields of `WDefInstances`  are not being updated. So this smells like this could be a larger RenameMap problem?

My advisors tell me I'm supposed to be graduating, so i'm punting to the pros on this one.

seldridge, 2021-02-09:
This is a bug. However, I think this should be fixed on master by b2726dff74847ccebc388b14330dcd54ddcc9895. I think you're right in that the issue is that the types aren't being updated. I didn't dig here, but `ManipulateNames`, if it doesn't update the types, then may cause bad Verilog emission later if something is depending on the types to set instance names.

@sequencer may be able to shed some light here.

Grabbing the latest snapshot with: 
```
cs bootstrap -r sonatype:snapshot edu.berkeley.cs::firrtl:1.5-SNAPSHOT -M firrtl.stage.FirrtlMain -f -o firrtl-1.5-SNAPSHOT
```

Compiling the above (as CHIRRTL output from Chisel):
```scala
circuit Parent :
  module Child :
    input a : UInt<1>
    output time : UInt<1>
    time <= a @[main.scala 12:8]
  module Parent :
    input a : UInt<1>
    output time : UInt<1>
    inst child of Child @[main.scala 18:21]
    child.a <= a @[main.scala 19:11]
    time <= child.time @[main.scala 20:8]
```

```bash
firrtl-1.5-SNAPSHOT -i Parent.fir && cat Parent.v
```

```verilog
module Child(
  input   a,
  output  time_
);
  assign time_ = a; // @[main.scala 12:8]
endmodule
module Parent(
  input   a,
  output  time_
);
  wire  child_a; // @[main.scala 18:21]
  wire  child_time_; // @[main.scala 18:21]
  Child child ( // @[main.scala 18:21]
    .a(child_a),
    .time_(child_time_)
  );
  assign time_ = child_time_; // @[main.scala 20:8]
  assign child_a = a; // @[main.scala 19:11]
endmodule
```

I can reproduce the error on 1.4.2, though.

seldridge, 2021-02-09:
This should be fixed via #2079 and will show up in the next minor release (1.4.3) Feel free to re-open if I missed something here.

closed 2021-02-09 by seldridge

davidbiancolin, 2021-02-10:
Perfect, thanks for the bp @seldridge. I should have searched more carefully for a previous issue.