chipsalliance/firrtl #2084: Remove all formal related verilog code when generate pure verilog (not systemVerilog)

Martoni, 2021-02-18:
### What is the current behavior?

As explained on this[ chisel-formal issue](https://github.com/tdb-alcorn/chisel-formal/issues/12), not all formal specific's code are removed when we generate verilog file.
This cause some problems when we try to simulate/synthesize the code. 

### What is the expected behavior?

As [the remover](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/transforms/formal/RemoveVerificationStatements.scala) delete assert()/assume() and cover() statement, is it possible to modify it to delete also this code :

```Verilog
  wire  ResetCounter_clock; // @[Formal.scala 14:36]
  wire  ResetCounter_reset; // @[Formal.scala 14:36]
  wire [31:0] ResetCounter_numResets; // @[Formal.scala 14:36]
  wire [31:0] ResetCounter_timeSinceReset; // @[Formal.scala 14:36]
  wire [31:0] TestCase_testCase; // @[Formal.scala 20:32]
  ResetCounter ResetCounter ( // @[Formal.scala 14:36]
    .clock(ResetCounter_clock),
    .reset(ResetCounter_reset),
    .numResets(ResetCounter_numResets),
    .timeSinceReset(ResetCounter_timeSinceReset)
  );
  TestCase TestCase ( // @[Formal.scala 20:32]
    .testCase(TestCase_testCase)
  );
```

### Your environment

- Chisel Versions: 
```
      "edu.berkeley.cs" %% "chisel3" % "3.4.2",
      "edu.berkeley.cs" %% "chiseltest" % "0.3.2" % "test",
      "edu.berkeley.cs" %% "chisel-formal" % "0.1
```
- OS: Linux  #46-Ubuntu SMP Thu Jan 7 00:51:04 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- Verilator version: 

### External Information

ekiwi, 2021-02-18:
Hi @Martoni, thanks for reporting this issue.

Unfortunately I don't think that there is anything we can do about this in the firrtl compiler: Since the firrtl compiler does not know anything about your Verilog blackboxes (since it isn't even able to parse Verilog), it cannot justify removing them, even if their outputs are never used.

I believe this issue needs to be solved by `chisel-formal`. Tom might need to add a custom pass to be able to remove his verification specific modules.

In my own prototype formal verification library ([kiwi-formal](https://github.com/ekiwi/kiwi-formal)), I circumvented this problem by [declaring all formal specifications _outside_ of the RTL Module](https://github.com/ekiwi/kiwi-formal/blob/d9be8efab5e8b7c7f04410ca943299b441bf1dad/test/kiwi/formal/StateMachineTest.scala#L50) You then need to [explicitly attach the spec when doing formal verification](https://github.com/ekiwi/kiwi-formal/blob/d9be8efab5e8b7c7f04410ca943299b441bf1dad/test/kiwi/formal/StateMachineTest.scala#L88). When you generate Verilog you just leave out the annotation that attaches the spec and thus you will just get the plain RTL from your Chisel module.
This approach requires some more typing though, so not sure if it is actually a good idea.

jackkoenig, 2021-02-18:
FIRRTL does actually have the ability to Dead Code Eliminate BlackBoxes (arguably a "Grey Box" in this instance): https://github.com/chipsalliance/firrtl/blob/edb91f7bc613026f824519786c3ce25740bb21c3/src/main/scala/firrtl/transforms/OptimizationAnnotations.scala#L64. I'm not sure if that's sufficient in this case. In the past we've discussed allowing BlackBoxes to have FIRRTL pseudo-implementations for purposes of optimizations.

Martoni, 2021-02-19:
Hi @ekiwi, @jackkoenig 

> I believe this issue needs to be solved by chisel-formal. Tom might need to add a custom pass to be able to remove his verification specific modules.

Yes, I thought so too. But when I saw the RemoveVerificationStatements.scala module, I thought it should be added here.

> In my own prototype formal verification library (kiwi-formal), I circumvented this problem by declaring all formal specifications outside of the RTL Module You then need to explicitly attach the [...]
This approach requires some more typing though, so not sure if it is actually a good idea.

This may be the solution: extend formal outside of main module.

```Scala
/* The module */
class KeepMax(width: Int) extends Module {
//...
}

/* The module with formal properties */
class KeepMaxFormal extends KeepMax with Formal {
//...
  // Some properties
  past(io.out, 1) (pastIoOut => {
    assert(io.out >= pastIoOut)
  })
//...
}
```

If we generate verilog from KeepMax we will avoid Formal generated code.

I'm going to test this.

Martoni, 2021-02-19:
It works. So it solved my problem. Thanks for your time.

closed 2021-02-19 by Martoni