chipsalliance/firrtl #2089: Source-level debugging

Kuree, 2021-02-22:
### Checklist

- [x] Did you write out a description of the feature you want to see?
- [x] Did you look around for any related features?
- [x] Did you specify relevant external information?

### Feature Description

Debugging generated RTL from Firrtl with existing simulator environment is less appealing since there are layers of optimizations that obfuscates the RTL. Ideally users should write their design in Chisel and debug in the same environment, e.g. setting breakpoint at a particular line in Chisel and inspect the frame/context information once the breakpoint hits.

We have been working on a source-level debugging framework called [hgdb](https://github.com/Kuree/hgdb), which is language and framework independent. With existing debugging information in Firrtl, we've built a [proof-of-concept pass](https://github.com/Kuree/hgdb-firrtl) that extract the information from Firrtl and dump a symbol table for debugging. The figure below shows the debugging experience in Visual Studio Code:

[image]

Our proof-of-concept has many limitations and we'd like to request the following features from Firrtl:
- Keeping track of symbol changes. Currently there is no symbol table or symbol tracking in Firrtl. We made some assumptions on how complex data types are lowered into RTL, but this may not work in the future, if Firrtl changes naming convention. This is particular helpful to debug SSA-transformed code, since we need the renaming map to faithfully represent the logic. We'd like to request such feature to keep track of naming changes and keep the symbol table updated.
- There is no frame/context information stored in Firrtl. As a result, once a breakpoint hits, users will not see any "local" value informations. This is not high priority, since all the named signals are represented in the "Generator Variable" sections. However, this is important to debug meta-programmed circuit since elaboration tends to destroy such information and makes the RTL confusing.
- `-O0` option. Because our pass only works on `High Form` IR before the mux pulling, a lots of symbols simply get optimized away when lowered to RTL, resulting in missing symbol when debugging the design. We'd like to request a flag that turn off all optimization passes, similar to `gcc`'s `-O0`.

### Type of Feature
 - new feature/API

### Related Features
None.

### External Information
hgdb is a second-generation of debugging framework we've been working on. You can check out more details here:
- hgdb: https://github.com/Kuree/hgdb/
- hgdb-firrtl: https://github.com/Kuree/hgdb-firrtl
- hgdb-debugger: https://github.com/Kuree/hgdb-debugger

Please let me know if you have any comments or suggestions.

ekiwi, 2021-02-22:
This is really cool work @Kuree!

I have one question: Since `hgdb` seems to be a step through debugger, how do you deal with the case when a connect affects a previously computed value? Like in your vending machine example, `incValue` is connected on lines 21-22, but used on line 18.
This is something that you would never have to deal with in most GP programming languages.

Some thought on how the current limitations could be overcome:

>     * Keeping track of symbol changes.

You could try and just make an annotation for every symbol that you can find in the HighForm. Annotations automatically get renamed, when the firrtl compiler renames things or lowers the types.

>     * There is no frame/context information stored in Firrtl

Can you give us an example of what you are looking for here? In firrtl we have a module level namespace, so you cannot shadow any variables inside a when context.

>     * `-O0` option.

If you chose the option of annotating all symbols, you can add a `DoNotTouch` attribute to your annotation which will ensure that the symbol is maintained throughout the lowering.

Kuree, 2021-02-22:
Thank you for your quick response @ekiwi!
> Since hgdb seems to be a step through debugger, how do you deal with the case when a connect affects a previously computed value? Like in your vending machine example, incValue is connected on lines 21-22, but used on line 18.
This is something that you would never have to deal with in most GP programming languages.

This is a good question and we've put lots of thought on that. `hgdb` itself doesn't care about the semantics of connected value: it just faithfully represents the information from the symbol table. In other words, the compiler needs to decide the execution semantics. For instance, if the logics are computed in sequence in a code block (think of `always_comb`), then it is natural to remap the symbols for each line. Consider the following example in some generic language:
```
a := 0;
if (b)
    a := 1
    a := 2; // <- target line
```
If compiler allows to break at target line, it needs to remap `a` to 1 in the symbol table to represent the execution order. This remapping can be achieved via standard SSA.
In the Chisel vending machine example, `incValue` should be mapped to the actual RTL signal `incValue` at line 18. However, since there is an implicit execution order at line 21-22. `incValue` should be mapped to corresponding SSA value. This also brings *another* question: how to access Chisel's SSA remap table? At Firrtl level the symbols are already mapped to `_GEN_` signals and thus lost their original information.

> Can you give us an example of what you are looking for here? In firrtl we have a module level namespace, so you cannot shadow any variables inside a when context.

Suppose we have a circuit designed in some generic language:
```
a := array[10]
for i in 0..10
    a[i] := i    // <- target line
```
In Firrtl, the loop body is very likely represented as unrolled 10 connection statements. Suppose we want to set a conditional breakpoint at the targeted line, e.g. `i == 2`. In this case, `hgdb` needs to know the frame information to evaluate the condition. It doesn't seem like Firrtl is storing such information. Maybe Chisel should capture it and store it somewhere?

> You could try and just make an annotation for every symbol that you can find in the HighForm. Annotations automatically get renamed, when the firrtl compiler renames things or lowers the types.

> If you chose the option of annotating all symbols, you can add a DoNotTouch attribute to your annotation which will ensure that the symbol is maintained throughout the lowering.

I'm not very familiar with annotations but I can give it a try.

ekiwi, 2021-02-22:
> In Firrtl, the loop body is very likely represented as unrolled 10 connection statements. Suppose we want to set a conditional breakpoint at the targeted line, e.g. `i == 2`. In this case, `hgdb` needs to know the frame information to evaluate the condition

I guess in Chisel, `i` would be a scala variable unknown to Chisel or firrtl. So I guess you would somehow have to hook into a Scala/JVM debugger to get that information.

> I'm not very familiar with annotations but I can give it a try.

We should be able to help you a bit with that. It might just take a little while since of course everyone is always busy.

For the other features like the SSA table and the frame info, it might be helpful if you could show us how this is implemented for other HCLs that you are familiar with.

Kuree, 2021-02-23:
> I guess in Chisel, `i` would be a scala variable unknown to Chisel or firrtl. So I guess you would somehow have to hook into a Scala/JVM debugger to get that information.

My educated guess is that it might be more difficult to get caller frame information from compiled languages than other scripting languages. I've had luck with JDI on Java before. I can hack around Chisel to see if I can get JDI working. In scripting languages like Python, getting frame information is much easier since the language supports it natively.

> We should be able to help you a bit with that. It might just take a little while since of course everyone is always busy.

Thanks!. Really appreciate that.

> For the other features like the SSA table and the frame info, it might be helpful if you could show us how this is implemented for other HCLs that you are familiar with.

Here is some sketch code used in the HCL framework that is co-developed with `hgdb` to test out capabilities and ideas. Since the HCL is a  DSL embedded in Python, getting frame information is much easier.

```Python
import inspect

def create_reg(name: str):
    # inspect caller frame info
    caller_frame = inspect.current_frame().f_back
    caller_locals = caller_frame.f_locals
    # store caller local variables
    store_local_(caller_locals)
    # actually create a register
    return create_reg(name)

# create registers in a loop
regs = []
for i in range(1):
    regs.append(create_reg(f"reg_{i}"))
```

In this case, whenever `create_reg()` is called, we will inspect the caller frame and obtain all local variables. We can then selectively store variables into our symbol table.

There is [a detailed explanation](https://github.com/Kuree/hgdb/blob/master/docs/README.md#context-scope-information) on how SSA remap table is used for hgdb. Here is the TL;DR; version

Suppose we have the following code in Python (it shouldn't matter in what language: the code itself can even be legal SystemVerilog.)
```Python
# a, b, c are variables that can be lowered to logic in SV
if a == 1:
    b = 0
else:
   b = 2  # <- line 1
c = b     # <- line 2
```
While we're transforming the code in SSA, it's common to create temp variables to hold intermediate values. Here is some example output in SV:

```SystemVerilog
logic[31:0] a, b, c, b_0, b_1, b_2
b_0 = 0;
b_1 = 1;
b_2 = a == 1? : b_0: b_1;
assign b = b_2;
assign c = b_2;
```
At the end of traversing the control flow graph at line 1, symbol `b` in SSA table will be mapped to `b_2`. As a result, we need to store the new mapping `b -> b_2` for breakpoint information at line 2. In other words, once a breakpoint hits at line 2, if we use the signal value of `b_2` to represent `b`, we will get result *as if* the code is sequentially executed, even though it's a continuous assignment under the hood.

ekiwi, 2021-02-23:
> Here is some sketch code used in the HCL framework that is co-developed with `hgdb` to test out capabilities and ideas. Since the HCL is a DSL embedded in Python, getting frame information is much easier.

Thanks for the example. This makes a lot more sense to me now. I believe that Chisel already does some parsing of stack frames. So you might be able to hook into that. Try checking out the `chisel3` repository and grep for `Stack` or `StackTrace`.

> Suppose we have the following code in Python (it shouldn't matter in what language: the code itself can even be legal SystemVerilog.)

It seems like in your language assignment to variables behave like blocking assignments in Verilog, whereas in Chisel/firrtl, connecting to a Wire or Reg is more akin to a non-blocking assignment in Verilog. Thus the "remapped" versions of `b` are never observed, besides the final version (in your example `b_2`). As far as I can tell, the `ExpandWhens` pass only ever calculates that final version for each signal. Thus I am not sure if a SSA remap table would make sense for `firrtl`.

An example to illustrate what I mean:

```
# a, b, c are variables that can be lowered to logic in SV
b = 1
if a == 1:
    b = 0
c = b
if a != 0:
   b = 2
```

In this case the value of `c` will be either `1` or `0`  (assuming that `a` is a 1-bit value).

However in firrtl with our "last-connect" semantics (meaning whichever connect happens last wins) you would get:

```
# a, b, c are variables that can be lowered to logic in SV
b <= 1
when a == 1:
    b <= 0
c <= b
when a != 0:
   b <= 2
```

Now the value of `c` is either `0` or `2` (assuming that `a` is a 1-bit value).

I am not exactly sure how that is going to affect `hgdb` but I wanted to make sure you understand the `firrtl` last-connect semantics.

Kuree, 2021-02-24:
> So you might be able to hook into that. Try checking out the chisel3 repository and grep for Stack or StackTrace.

Thanks. I will take a look.

> I am not exactly sure how that is going to affect hgdb but I wanted to make sure you understand the firrtl last-connect semantics.

In that case it is much simpler since there is no ordering involved, and it is supported by `hgdb`, since it behaves like a non-blocking assignment. That also implies that step-over based debugging is less effective in Chisel, as each statement doesn't have much side-effect.

That being said, IMO it might still be useful to use step-through debugger to figure out the conditional execution, e.g. `when` construct; that's part of the reason why people step over their code anyway.

mwachs5, 2021-10-06:
Really cool @Kuree ! are you presenting your work anywhere in the near future?