chipsalliance/firrtl #2106: What are the rules for naming temporary wires?

itviewer, 2021-03-05:
I am working on visualizing the firrtl circuit and want to remove the temporary wires. 
I thought that the temporary wires started with _T or _GEN, but I found this situation
```
    node _x_T = sub(x, y) @[GCD.scala 28:24]
    node _x_T_1 = tail(_x_T, 1) @[GCD.scala 28:24]
```
I want to know if there are any rules for the naming of temporary wires, is it true that strings containing _T or _GEN are temporary wires?

mwachs5, 2021-03-05:
Hi @itviewer are you starting from Chisel to generate your firrtl then verilog? You may find this article helpful: https://www.chisel-lang.org/chisel3/docs/explanations/naming.html

jackkoenig, 2021-03-05:
I wouldn't say there are formal rules, but the general rule of thumb is that signals beginning with `_` are temporaries or "unnamed signals".

itviewer, 2021-03-06:
@jackkoenig 
 
I think I should treat all node statements as temporary and replace them with the basic operations on the right side.
There seems to be no problem with this method, You can see the rendering effect [here](https://www.easysoc.org/icviewer/?graph=N4IgTgpgbglgzjA9gOxALgAwBoQBcCeADhOiAOZgCGhAFiDjACakXV04DGNMANo5KjQBtUHAg8IHXBGZoAZpR5ichRAlxJBoAB7oAjADYATADpsIfOgDMVswF8cCAF4k0oAO5NcdNFYCsAJxmODQQMGQ0uOgALADsfvY4BMSkyIiMJAyy5FS0JgDiAMIAIvQgXLz8EIIieBDaUWggRaWO4pLSsgpKECpqMBoo6Dr6BgbBFugAtEZ6iSDOrh5ePkaxE6HhkaPzya4gPJQARuJlTCy5NAUlJocnSgDEGCAOou1SMujdyiCq6prDEC6NBTBLmSxoAy2PwOBYwFyAzyMbzoMEhMIRRpg2F7UiqMBRLIXNjXYomDg8RAcADWZQqfAE6Fq0gapApVNpbQkHy6ih+fwGALcQOm0XG4PQ4thi0RKxiaJAm0xOwwOKI+zupyJTVYeRa5MpNNux3EcCeLwAuq8Fu9Ol8+b1fv1BloRSCFRCAgkYY54UsQEiUWgFUrtsHduq8YgCWdsrqrvrIGJCeVuAzqkzQCzGuAIMmymJuXb5A6+v8hsLgVMxRMIVLfQjhYGfNEQxiw4YIykmpqeLHiXqbkmILhjfczc87FasG8i58Sz0y4KKyN3bXrHMfXDG8tkT421ssV39viU+cdZdSSYkAB9KCKACuED0dLTVRqWfqOdv954T5fXIdPO3yOgKLqAlWAAcpgSpC8wyk2cpoJBczmKGjSdqqSSRj2Jp9tqOQkvqP6Ps+Y6muaU7WoWQG8ouTrlq6VYetYVjrFh27+s2qIbO2R4cbiTSnv2F5ETcJF-hARivpUjLCJ+rJNBJT7SYBPL2vRYFCquUzQeucEcQhu5BihvGHiqardgceEiYRg5kspUnkY8k7TrOtEafyzraW6oL6YY4xbkZAZIQeyrhgJOFOjGBHxlet6UpQjAwMgZAAGqkXAMnph+dSKSACWIElKXpZlBa2sBpYMcuTHTHorb6fWnGynu+jRNEZnhZhlkajZsWXsRiA3olyWpRlklwM5E6WtRFV0V5jEQdYgT6UYGAmEFfotUGYVhti2FWcJ-ViQ5Q2IA+uCEBdLTZe+mZ5d+Z0XVduA3WpxYgUu4GVpK+lTHo7HSltiGtWgegYJBnUdk1gnWfctlxYNN7nZd103L201UTONpzvNoHeSubr+EEsFGEYTXBdxEXouZEU9VGMUFXGA3iU9qO4ON5ycG+cnMl+pC3ijL2c8w72VZpBO1QZOB1vBwPGT4ejk1DGEw1FvYIyzp3I89F0i1NlFuTjHkLgtNVLWgcSNRT8shaDVjrWhfExI79NNGkGSaydJi6NzskZvJD2kL7xvqab+OLT9YPrX4MvTLMcs7nbQYkzTXVq1ZGvHfZPsG5O2PZqQkBkOVuOeRH5tRyxYMwUDSdU2sKsWQdvXw9nCY3NoBsvrNZfh19PnMf56ybfXoVN3TLcM2ezPe13KUvNjNFh591XfauDvDxtddcePTu0-teBRUdTMDh3ZJdyjM1L3N5cD4TwJW6TGCoTv20+Jv+-hdErtT+76SZFPqJHOlg-Y5XuoXJooDQ4fSqlpB++gY5xxBAnQytsqap0VM7MGGdW5aiAXZc+Jh8B50XgpHMxdS4m1XvAqW1claJ13qDRuX9obHlIFnAhiMbgkIxg8HuN8+40MlhbPysEAbbwbEwnaE9D6wxPueQhV4SEL17tQuBIio6f2QRI0e0j9yyPYUJaMM8z7KJMFfLG7kV4aMjquPQEjSaQSCHo9+6A1oTx-kYkAHtAGKO4WSMguAbwAVTP7XKkDyApmXrAiWdi3SoWQTMV+Ui3Fg0hqw1W3jOH+K1iYIJITSFqJsXEyuOl6G-2aiDGRmSeKRUOiYr2OcClzBSgI6xsSzbr18tXIwh9KZ7zTntbxCjZ7NOCa05A0limdIrt04EHjxEjzftUgxtTJ5HwaYzXJ3sWkWIutfDp4sumD30I45BsRoQrIVu4x2QzGhePqfsXxTSiFwAfEcG8qkwngMDpE95RwqElJOQgkE0IkmoOucnVY7F7nN02Xg-CXC8kAq+UUwR6jSnzOmBUxhaTdr8TdtFUxwC3kfLRW0shMDjlzNOWuUm-T0GDKwQfEZjT25XlRaYFK0yMXAtpaCxZOjlmpNWXUuFGz5HsuRd7Ll+yohWOpXje+UtDCwsmMhFJVSbloCFSy7+lTYYvI5fqXAlBeA3isLdXm5DSBmt4EC2ZKqLZ3I1ckvFYrfDigld1P+cN8E7Jzvanglr0VHOVWvOlYidGVIGaDAldSiWjLMaa81IbbCUpmTS51Udo0ar6R6nVCbJXH2lYGohwbQ08utQHPm+VQkxOzZG0F3q3XeihRgievqEUcL6jKoNaaq1TKKUbRtEbaEWz1RCXRHbmXoUTX65NpKryVtsJY0dt9+7NtVecjVVg9KuM9Xq+dltDVRWNf2slnzog1oifzJoALHVNonbm8FbrIWip1Swn1uDe1t0vZy8lHU+H53DXfbdojemMrHvGwxTzp6vMA9e68yB2lKvAy+8p-lY1Mtg+suRpbtljKvTeDq1as3js0auKd+gRXauheK-Vwz4PGKIymm4qKOrrooxhqjCSDDqohFYAIOGYNBmPdgx5RKL3lpXYOvwt6IH3rwGmp9lH4nAldRCd1aCxMf1bSe7tsMcnEbkxahIIGqVjt4xpnF2HC0Mepkxwli6y2mdTeZlDaHrNbswz01a0H9GMZPQRrZJKlEeZDQkatYC7p-OUw2zdwjbPSzbZI+jnb1lGfVn22TkWbzReHZZxVPnktlLdDRsGdG401IlaFk8bn2NkkrQkbjfKnUQajmTDJMbbC6aC74GCdXf3-09iam4ABbB82gQmKfi-lKbIdSu2PKwsjaEKtU1ZhV2kb-qkV5cm9NwpxX2vPr40PWCXj4izrw3VtlbHl36kW4UmJc2605kLGpmzq3fqwT8oer96rDO7ZM01kwz25gxJHTx3z527MMvS1t4L2D6sIfG2SCHKHvn0ji+9gWUyvuw5S627T7bP2Oe-c5+FxncvucOzNyZphisbqESt7F9KY0mD6Td2rVOS1hcQ09o7kzQk45tUHJSqHCdlfZyTuzAOKdA+wdlzOtOweY7adD076mfuDYC4j3DvOQv3fCwE8Hwv5WHPQ0T3X3X9L7truTqmVght878LtmTdOMdHexzzWttqmiLel2zulphY7vs24b7bWWQdq8e-TtFJ2wM2-Z3miEV2FdU2Laj1jpu8nPdMK92L4v-n4OW1iulcu1yZ6QpT4H2S48RYT4X20Wvk8y6jVBg3enkespY8SwXzesdvYDwVAnYsdey7+mTjLtele0xV4iwf3uZvcqK3hTGLPMUgqlmnxBXOa+3bdyb5f5vV9eZHxLsfiXWcV5bdP7vA26-K9j-+g7K+KWobb9bjvoKXf68PyNxRxP3RzPzRTa3bxD1BQcUE1uUhkANWFdRPSkz9U9zBwgAAEcbwDBL9IlMDg879VU-oP1Z9QY9Aes+dF8-0A0vcTBMDsCw0f8oDd9cV+t8U4Mk1Gt48yR6DxhM1tdvtU8u82DPVs8QCAN9ReDh9i9-cr9ngJ9BDK8H8ED0BMF69+9QduC6CsC+D19xxDYYdf8pZKsZ0nc51gD+8l0m8eCdDLcSsktmDJ0RNsM+secP5XcQsPcAFT9nsrUZC70FtpsCCd9J11sI8HMG55904G839aDfDGDy8Qjc1q4M83De9woc8B9QD4ii8flcdR9PsFCU8lC-sZ8kddUoi2ENDG8zccjW9mdDDHDkiAC0inNjdLCuDrCwCM1x88iS9lMYsmDCCLYq9-tWjn8F9X8aD1cjseimcN8DCBDijQU99o4D9WixCOiHsuj4jKV-ClN8o9ihikjVxRiyio93FKislqjYiZiZsei5gGilijCLZ-8EcVC2iLDODtjajZi7Ct9+Uc1qNnDYIHcIikI3jhtvE0CtDnsb19j5scwg8iiXiuswjtMSDyiJjoibjpjYSjtgMFjQNjiBUWD9JUizCj92jvi89vY4STBcixdZDS8kVEjSSRjlDxirjqcctbj8SZsOoocnjIDhjmj3iNiODXMfj88CTpC+jmSBjei2SgS3QzjH80lsSqiiVNCdjZSeVv9lTOssNxF1pudKSgC+8aSfC9SpcES8dJcb9t92So41SPjNTrjtSaiZSBSvMDSHDRSN5XcIQC0PjNirTsjZSICFhzowAOAIAABJOIv4q+JISgMAMgEcRMsHeeVAHAMAFGEqAABUQBSlwCykDmBBd1gLBisBSQ3jYn8lrMSHrOrIYSwkDP8kdw7NgkgnrFXGiCDPQF7PsAtBRKqlhhkAzNP2fG8xjLjKzK0MvgOVTPTMzNoMFl1lehKDKHzIuiLJLOQDLPumBD0h7L7ISQwEHOQnPOBHBivKMFcOxkrIMA6lJlcNHJJL9UnL8VoOfF5VEDnITNoJIRTJU1XNwAXK6M4wvzzILNSmLNLPLNqBPPvI-VXACFQrrLdAwvJPBmbISTbOQWiDwtVA-MNInMYCnNAOfD8PlNykNL82BBwtJhINhgACsHxkApAARqK9ArAbw2LXJqJALILfiGdLcVyMyIKkz7i5TdyNB4KDyjyKz3EDB6Emz2yKtIJ1LX4yL-SfgKKqKJCbhnwb1hKHxYygL0DbDQKzVwLRK8kNz2YRYdy4KyAELDykL7EBzVojB4DrRnzXzkE7cYQ9Lb8DKopvzpy9AFNzLLKHLvZCpioxoypJK1y7jSMGTbRXK9zFLELjz4cdEDASYArrArzDASqnyyryS-AsKFl4gaq6rLiUiC0zTqNtLyTWrQqxz6JDKfz0DDBL8GK4c9dxFirvEOKuLwJeKDABKhKl4RLgKJKwKpKEqc4C8YLwA3KPLlLkKhzML8KmKDrNKjr7c-AY5SrdU+KzqLqqqrq+tkFhMOprs7qyYHq90AhnrurPye0mgoreLYghr9LnT0Krz-BbrfqQBJruKhgAa5qrc4BFqwc5VbK0zVraDV1Nr5L9z8qVKwYfLYIAgrk7rDAryibt4wqnSvzKL+qtDnxIIrMkatCQLlyVr0qtC5VBjsa8rPKCrkJjrLrmLgq0LsLyqX5DqzkxbdKeqIqrJ-rjKeC9AAhGaLL5ylrUb7LaC9kubtqlKvK3RTzhamq0Ahb80RbTrQTHFLqHEwarbKbATIb5b386C1ogbwqQbRb7cra-VobpqFaXaMB4aSsmauiWaUw7L0brKGCjjub [... cut ...]
    node _y_T_1 = tail(_y_T, 1) @[GCD.scala 29:25]
    node _GEN_0 = mux(_T, _x_T_1, x) @[GCD.scala 28:15 GCD.scala 28:19 GCD.scala 25:15]
    node _GEN_1 = mux(_T, y, _y_T_1) @[GCD.scala 28:15 GCD.scala 26:15 GCD.scala 29:20]
    io_outputGCD <= x @[GCD.scala 36:16]
    io_outputValid <= eq(y, UInt<16>("h0")) @[GCD.scala 37:23]
    x <= mux(io_loadingValues, io_value1, _GEN_0) @[GCD.scala 31:26 GCD.scala 32:7]
    y <= mux(io_loadingValues, io_value2, _GEN_1) @[GCD.scala 31:26 GCD.scala 33:7]
```

closed 2021-03-06 by itviewer