chipsalliance/firrtl #2114: readmem block is not picked up by synthesis tools

carlosedp, 2021-03-11:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?

In the recently implemented inline readmem annotation, in some cases synthesis tools do not pick-up the readmem statement when placed inside the `ifndef SYNTHESYS` like it currently is.

```verilog
...
`ifndef SYNTHESIS
`ifdef FIRRTL_BEFORE_INITIAL
`FIRRTL_BEFORE_INITIAL
`endif
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  ram_MPORT_addr_pipe_0 = _RAND_0[10:0];
  _RAND_1 = {1{`RANDOM}};
  ram_MPORT_1_addr_pipe_0 = _RAND_1[10:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
  $readmemh("insns.hex", ram);
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule
```

Verilator for example works correctly but Yosys does not.

### What is the expected behavior?

The following Verilog, works correctly with the readmem outside the `ifndef`. A new `initial begin ... end` pair must be used.

```verilog
...
`ifndef SYNTHESIS
`ifdef FIRRTL_BEFORE_INITIAL
`FIRRTL_BEFORE_INITIAL
`endif
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  ram_MPORT_addr_pipe_0 = _RAND_0[10:0];
  _RAND_1 = {1{`RANDOM}};
  ram_MPORT_1_addr_pipe_0 = _RAND_1[10:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
initial begin
  $readmemh("insns.hex", ram);
end
endmodule
```

### Steps to Reproduce

Create a simple memory module with the load annotation.
Check that the synthesis tool does not load the memory from file.

I'm already looking into providing a fix.

ekiwi, 2021-03-11:
In yosys, does it work if you do?:
```
read_verilog my_mem.v -DSYNTHESIS
```

jackkoenig, 2021-03-11:
This is a bit tricky because as @ekiwi is alluding too, `$readmemh` isn't synthesizable in ASICs (nor in some FPGAs).

ekiwi, 2021-03-11:
I misread the if**n**def as ifdef.

So yeah, what is happening is - as Jack pointed out - for ASIC tools we do not want to have any memory initialization (which is strictly speaking non-synthesizable anyways). However, yosys seems to implicitly define `SYNTHESIS` so what you would actually want is for `yosys` to **un**define `SYNTHESIS`.

ekiwi, 2021-03-11:
You could - for now - work around this problem by using `-formal` on the [read_verilog](http://www.clifford.at/yosys/cmd_read_verilog.html) command.

carlosedp, 2021-03-12:
Confirming that using `-DSYNTHESYS` does not work but `-formal` works since it replaces the SYNTHESYS define.

I understand that the SYNTHESYS define is used for ASIC as well, that's why the memory initialization is in that `ifndef` block right?

Taking the readmem out of that would break the premise that initialization shouldn't be present for ASICs for example.

Any idea on a strategy that would be acceptable for both FPGA and ASIC worlds?

carlosedp, 2021-03-12:
Just tested one project (Chiselwatt, an OpenPOWER core) synthesized with `-formal` and it didn't worked with the readmem statement inside SYNTHESIZE `ifndef`. 
Outside that ifndef it works out-of-the-box.

ekiwi, 2021-03-12:
We have put an item on the chisel dev meeting agenda for Monday. We might consider moving the memory initialization outside of the `ifndef SYNTHESIS` block when the `-target:fpga` flag is set.

jackkoenig, 2021-03-12:
I think handling this via `-target:fpga` is probably the most pragmatic solution, but I don't love it. It violates the principle of writing reusable hardware generators; if you use this feature for functionality, a design that works on an FPGA won't work on an ASIC. That being said, it's hard to solve this elegantly (I'm thinking something like https://github.com/chipsalliance/firrtl/pull/1050), and it's better to solve it pragmatically than not at all.

carlosedp, 2021-03-12:
Yes, that's what I though. Moving it out of the ifndef when the target is FPGA. :)
I look forward to join, thanks for the invite!

carlosedp, 2021-03-12:
> I think handling this via `-target:fpga` is probably the most pragmatic solution, ...

I agree and would love to not worry about this but to what I understand there is no "one size fits all" here since both FPGA and ASIC generation goes thru the ifndef in different ways.

jackkoenig, 2021-03-12:
To be clear, my proposal here is longer term, not short term. I'm on board with `-target:fpga` making this work in the immediate term.

> I agree and would love to not worry about this but to what I understand there is no "one size fits all" here since both FPGA and ASIC generation goes thru the ifndef in different ways.

While there's certainly not "one size fits all," I think we *can* do something general here. Just like in #1050 where we support FPGA preset by transforming a register with an async reset into a register with no reset and a preset in the Verilog. I can imagine doing something similar with memories--associate an async reset signal and when using ASIC emission we transform the memory into flops with an async reset. For larger memories, we could also keep the memory but add a state machine for a multi-cycle reset. The latter is slightly problematic because you need to make sure you don't use the register until reset is done which has logic implications which is why this is a bit tricky. Anyway, food for thought for future work 🙂

ekiwi, 2021-03-12:
>  I can imagine doing something similar with memories--associate an async reset signal and when using ASIC emission we transform the memory into flops with an async reset. For larger memories, we could also keep the memory but add a state machine for a multi-cycle reset.

You could do this without changing anything else by adding a transform for memories that have a MemoryInit annotation.

jackkoenig, 2021-03-12:
> You could do this without changing anything else by adding a transform for memories that have a MemoryInit annotation.

Sort of, the problem is you need to know what reset signal to use.

ekiwi, 2021-03-12:
> Sort of, the problem is you need to know what reset signal to use.

Oh yeah, adding a reset to each memory port would be interesting and would allow us to make some better guarantees for output values of SyncReadMems after reset.

PR #2166 (2021-04-01): Add memory initialization options for synthesis
This PR adds options for memory initialization inside or outside the
`ifndef SYNTHESIS` block.

Fixes #2114.

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [x] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you specify the type of improvement?
- [x] Did you state the API impact?
- [x] Did you specify the code generation impact?
- [x] Did you request a des [... cut]
closed 2021-04-01 by jackkoenig