chipsalliance/firrtl #2129: crash 2651

drom, 2021-03-17:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?

The following FIRRTL program.

```python
; seed: 2651
circuit top_mod :
  module top_mod :
    input clock: Clock
    wire _tmp18: SInt<2>
    wire _tmp27: SInt<24>
    _tmp18 <= rem(asSInt(UInt<21>(932719)), SInt<2>("b-10"))
    _tmp27 <= add(_tmp18, SInt<23>("o-15035672"))
    assert(clock, xorr(UInt<16>("h4045")), xorr(_tmp27), "assert31")
```

Running with `firrtl-1.5.0-SNAPSHOT` crashes with the following error:

```
firrtl
Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! trying to emit unsupported operator: UnknownType
Please file an issue at https://github.com/ucb-bar/firrtl/issues
        at firrtl.Utils$.error(Utils.scala:491)
        at firrtl.Utils$.throwInternalError(Utils.scala:164)
        at firrtl.VerilogEmitter.emitCol(VerilogEmitter.scala:205)
        at firrtl.VerilogEmitter.$anonfun$emitCol$2(VerilogEmitter.scala:197)
        at firrtl.VerilogEmitter.$anonfun$emitCol$2$adapted(VerilogEmitter.scala:196)
        at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:169)
        at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:165)
        at scala.collection.immutable.List.foldLeft(List.scala:79)
        at firrtl.VerilogEmitter.emitCol(VerilogEmitter.scala:196)
        at firrtl.VerilogEmitter.$anonfun$emitCol$2(VerilogEmitter.scala:197)
        at firrtl.VerilogEmitter.$anonfun$emitCol$2$adapted(VerilogEmitter.scala:196)
        at scala.collection.LinearSeqOps.foldLeft(LinearSeq.scala:169)
        at scala.collection.LinearSeqOps.foldLeft$(LinearSeq.scala:165)
        at scala.collection.immutable.List.foldLeft(List.scala:79)
        at firrtl.VerilogEmitter.emitCol(VerilogEmitter.scala:196)
        at firrtl.VerilogEmitter.emit(VerilogEmitter.scala:124)
        at firrtl.VerilogEmitter$VerilogRender.$anonfun$emit_streams$6(VerilogEmitter.scala:1106)
        at firrtl.VerilogEmitter$VerilogRender.$anonfun$emit_streams$6$adapted(VerilogEmitter.scala:1106)
        at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:563)
        at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:561)
        at scala.collection.AbstractIterable.foreach(Iterable.scala:919)
        at firrtl.VerilogEmitter$VerilogRender.emit_streams(VerilogEmitter.scala:1106)
        at firrtl.VerilogEmitter$VerilogRender.emit_verilog(VerilogEmitter.scala:1226)
        at firrtl.VerilogEmitter.$anonfun$emit$2(VerilogEmitter.scala:1267)
        at scala.collection.immutable.List.foreach(List.scala:333)
        at firrtl.VerilogEmitter.emit(VerilogEmitter.scala:1261)
        at firrtl.VerilogEmitter.$anonfun$execute$2(VerilogEmitter.scala:1279)
        at scala.collection.immutable.List.flatMap(List.scala:293)
        at scala.collection.immutable.List.flatMap(List.scala:79)
        at firrtl.VerilogEmitter.execute(VerilogEmitter.scala:1276)
        at firrtl.SystemVerilogEmitter.execute(SystemVerilogEmitter.scala:29)
        at firrtl.Transform.transform(Compiler.scala:319)
        at firrtl.Transform.transform$(Compiler.scala:319)
        at firrtl.SeqTransform.transform(Compiler.scala:401)
        at firrtl.stage.transforms.ExpandPrepares.execute(ExpandPrepares.scala:19)
        at firrtl.Transform.transform(Compiler.scala:319)
        at firrtl.Transform.transform$(Compiler.scala:319)
        at firrtl.stage.transforms.ExpandPrepares.transform(ExpandPrepares.scala:7)
        at firrtl.stage.transforms.CatchCustomTransformExceptions.execute(CatchCustomTransformExceptions.scala:10)
        at firrtl.Transform.transform(Compiler.scala:319)
[... 60 lines cut ...]
        at firrtl.options.StageMain.main(Stage.scala:71)
        at firrtl.stage.FirrtlMain.main(FirrtlStage.scala)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at coursier.bootstrap.launcher.a.a(Unknown Source)
        at coursier.bootstrap.launcher.Launcher.main(Unknown Source)
```

### What is the expected behavior?

Not to crash. Give a more informative Error or Warning or just work.

### Steps to Reproduce

`./firrtl-1.5.0-SNAPSHOT -i top_mod.fir`

### Your environment

`Linux linux-sydi 5.11.4-1-default #1 SMP Mon Mar 8 05:16:55 UTC 2021 (be77cd2) x86_64 x86_64 x86_64 GNU/Linux`

- Verilator version: `Verilator 4.109 devel rev v4.108-38-g36eb952b`

### External Information

seldridge, 2021-03-25:
This is reproducible on `master` (`1.5-SNAPSHOT`), but not on `1.4.2`. It looks like some transform is adding an `UnknownType` and should be invalidating `InferTypes`.

drom, 2021-04-20:
still crashing

itviewer, 2021-04-27:
I think https://github.com/chipsalliance/firrtl/issues/2129 https://github.com/chipsalliance/firrtl/issues/2127 https://github.com/chipsalliance/firrtl/issues/2126 are essentially the same problem caused by the following code, which does not consider the case where the literal is directly used as argument
@sequencer 

https://github.com/chipsalliance/firrtl/blob/54b6d8713bddb0e16e475a4f47e376217c5261e3/src/main/scala/firrtl/backends/verilog/VerilogEmitter.scala#L283

@drom But in fact, this is another question, should we use randomly generated Firrtl content to test this project?
The Firrtl content generated by Chisel naturally conforms to the FIRRTL Grammar, but the handwritten Firrtl content may not be consumed by this project.
In these examples, the Firrtl file directly uses SIntLiteral. Although these statements conforms to the Grammar, this phenomenon does not occur when the Firrtl is generated by Chisel (which use asSInt instead of SIntLiteral?). The FIRRTL parser can indeed parse the handwritten Firrlt content, but in some places, such as the processing of Verilog, only the necessary cases will be considered.

drom, 2021-04-27:
> @drom But in fact, this is another question, should we use randomly generated Firrtl content to test this project?
> The Firrtl content generated by Chisel naturally conforms to the FIRRTL Grammar, but the handwritten Firrtl content may not be consumed by this project.
> In these examples, the Firrtl content directly uses SIntLiteral. Although these statements conforms to the Grammar, this phenomenon does not occur when the Firrtl is generated by Chisel (which use asSInt instead of SIntLiteral?). The FIRRTL parser can indeed parse the handwritten Firrlt content, but in some places, such as the processing of Verilog, only the necessary cases will be considered.

let me see if I got this right @itviewer 
You are asking: if it is OK for FIRRTL to crash when legal FIRRTL program is compiled?
I don't know. It is the question to FIRRTL gurus. @jackkoenig @azidar @seldridge

itviewer, 2021-04-27:
@drom Sorry, it seems that the problem is not caused by SIntLiteral. Let me see if I can figure out what the problem is.

seldridge, 2021-04-27:
> You are asking: if it is OK for FIRRTL to crash when legal FIRRTL program is compiled?
> I don't know. It is the question to FIRRTL gurus. @jackkoenig @azidar @seldridge

😂 

You're both right in different ways. This is clearly a bug. If we crash on any legal input FIRRTL, it's bad.

If the code is legal FIRRTL that is _impossible_ for Chisel to ever emit, it's less worrying, lower priority, etc. However, I'm pretty sure somebody could generate this code in a transform and cause problems. The FIRRTL compiler should be able to stand alone.

If the bug is what I think here (a transform can produce `UnknownType`, but that transform does not list `InferTypes` as something it invalidates), then the best way to prevent this in the future would be for me or somebody to revive: https://github.com/chipsalliance/firrtl/tree/check-transforms. That automatically checks transforms to see if they violate any invalidations. In the past, problematic examples fall out quickly from that. As a quick hack, dumping the circuit (with type information) inside each transform in the Verilog emitter may provide some clues as to where this is going wrong.

PR #2485 (2022-03-02): Fold VerilogModulusCleanup into LegalizeVerilog
This fixes handling of signed modulus and removes some redundant work.

Fixes #2480, Fixes #2129, Fixes #2287, Fixes https://github.com/chipsalliance/chisel3/issues/2012

This has been a bug for a long time. I've marked it 1.4.x but even that might be a pain to merge due to conflicts with https://github.com/chipsalliance/firrtl/pull/2304

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [NA] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you add at least one test demonstrati [... cut]
closed 2022-03-02 by jackkoenig