chipsalliance/firrtl #2218: Aggressive simplification hiding uninitialized registers

johnsbrew, 2021-05-06:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant ~external~ information? ;-)

### What is the current behavior?
Aggressive simplification hiding an uninitialized register
https://scastie.scala-lang.org/KKD9BUW9SieFs7KtN3VeTQ
```scala
class Example extends MultiIOModule {
  val en = IO(Input(Bool())) 
  val out = IO(Output(Bool()))
  
  val r = Reg(Bool())
  when(en) {
    r := true.B
  }
  out := r
}
```
```verilog
module Example(
  input   clock,
  input   reset,
  input   en,
  output  out
);
  assign out = 1'h1; // @[main.scala 14:7]
endmodule
```

### What is the expected behavior?
(derived from https://scastie.scala-lang.org/VE7H5uCSTiGQxFXQUDoO8A)
(note : removing ifdef pour conciseness)
```verilog
module Example(
  input   clock,
  input   reset,
  input   en,
  output  out
);
  reg [31:0] _RAND_0;
  reg  r; // @[main.scala 11:14]
  assign out = r; // @[main.scala 15:7]
  always @(posedge clock) begin
    if (en) begin // @[main.scala 12:12]
      r <= 1'h1; // @[main.scala 13:7]
    end
  end
// Register and memory initialization
initial begin
  _RAND_0 = {1{`RANDOM}};
  r = _RAND_0[0:0];
end // initial
endmodule
```
### Your environment

- Chisel Versions: x.4.3
- OS: MacOS Darwin Kernel Version 17.7.0
- Verilator version: 4.200

### Additional experimentation
- works with a signal `in` instead of `true.B`: https://scastie.scala-lang.org/VE7H5uCSTiGQxFXQUDoO8A
- works with a `otherwise` block following: https://scastie.scala-lang.org/StT53SfnSe6mt67f4683hg

### Conclusion
Chasing uninitialized signals is often a difficult task for hardware developers, Chisel/FIRRTL stack (strongly oriented towards 2-states simulators, namely verilator & treadle) is dangerously making it even more difficult ...
Open question: how do you deal with this issue in your own designs? Don't say you have never seen any uninitialized register passing through synthesis, I won't believe you ;)
We are starting to include icarus as slow but 4-states simulator in our flow to mitigate this issue.

ekiwi, 2021-05-06:
## Const Prop + DCE

The optimization that the firrtl compiler does is correct and one might even argue that the fact that the register gets eliminated actually exposes the bug better, since by default (e.g. in `chiseltest`, see below) the register would start with zero and the design would behave as if the Chisel was correct.

Would you like to have an (optional) warning from the compiler when a register gets eliminated? There are some generators (afaik) that may depend on registers getting optimized, so we cannot really make this an error.

## Register Initialization

What does your testing setup with verilator look like if I may ask?

We did screw up in not randomizing registers by default in `chiseltest` which can hide a lot of errors. Fixing this is on my `TODO` list, but there are many higher priority items... (I would be happy to help with a PR though).

phhorrein, 2021-05-07:
I think this is really an issue. With this configuration, if I forget a "0" initialization, I will get a 1 at the start and it mighnt never be caught by my tests. 

Chiseltest is one thing, but if I generate this and integrate it in a wider project using QuestaSim (for example), seeing an "X" there is much easier to catch. If I don't catch the issue, the fuse behavior I wanted to get will be buggy. 

I really think that when doing this, FIRRTL is trying extract the intent of the developer.

phhorrein, 2021-05-07:
Regarding registier initialization, not randomizing registers seems to be a good practice, as Verilator is able to randomize "X" values by itself. And randomizing is an issue with other simulators, as for example I greatly prefer seeing an "X" on Icarus or on QuestaSim instead of a randomized value which a 1/2 chance of being correct (and thus, of not being caught by my tests).

cg31, 2021-05-22:
There is also a ticket about this at chisel3:
https://github.com/chipsalliance/chisel3/issues/1444

It is better to add an option to skip emitting code for randomizing registers.

albert-magyar, 2021-06-15:
Since this has somewhat shifted focus to the initialization concern, I'm going to close this for now. Currently, it's possible to avoid const prop of registers to refinements of the original FIRRTL behavior with a [NoConstantPropagationAnnotation.](https://www.chisel-lang.org/api/firrtl/latest/firrtl/transforms/NoConstantPropagationAnnotation$.html)

closed 2021-06-15 by albert-magyar