chipsalliance/firrtl #2282: can emitFirrtl but not emitVerilog

makslevental, 2021-06-23:
### What is the current behavior?

can run

```
object LineBuffer extends App {
  val dWidth = 16
  (new ChiselStage).emitFirrtl(
    new LineBuffer(dtype = UInt(dWidth.W), colsIn = 3, rowsOut = 3),
    args
  )
}
```

but 

```
object LineBuffer extends App {
  val dWidth = 16
  (new ChiselStage).emitVerilog(
    new LineBuffer(dtype = UInt(dWidth.W), colsIn = 3, rowsOut = 3),
    args
  )
}
```

gives me an exception

```
Elaborating design...
Done elaborating.
Exception in thread "main" firrtl.FirrtlInternalException: Internal Error! trying to emit unsupported operator: UnknownType
Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:491)
	at firrtl.Utils$.throwInternalError(Utils.scala:164)
	at firrtl.VerilogEmitter.emitCol(VerilogEmitter.scala:212)
	at firrtl.VerilogEmitter.$anonfun$emitCol$2(VerilogEmitter.scala:204)
	at firrtl.VerilogEmitter.$anonfun$emitCol$2$adapted(VerilogEmitter.scala:203)
	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
	at scala.collection.immutable.List.foldLeft(List.scala:91)
...
```

### What is the expected behavior?

obviously should be able to run both...

### Steps to Reproduce

is giving you guys the firrtl sufficient to repro?

```
;buildInfoPackage: chisel3, version: 3.4.3, scalaVersion: 2.12.12, sbtVersion: 1.3.10
circuit LineBuffer : 
  module RingBuffer : 
    input clock : Clock
    input reset : Reset
    output io : {flip inData : {valid : UInt<1>, bits : UInt<16>[3]}, outData : UInt<16>[1]}
    
    cmem ram : UInt<16>[3] @[RingBuffer.scala 28:16]
    node _outputPtr_T = eq(io.inData.valid, UInt<1>("h00")) @[RingBuffer.scala 29:32]
    reg outputPtr : UInt<2>, clock with : (reset => (reset, UInt<2>("h00"))) @[Counter.scala 60:40]
    wire outputPtr_wrap : UInt<1>
    outputPtr_wrap <= UInt<1>("h00")
    when _outputPtr_T : @[Counter.scala 118:17]
      node outputPtr_wrap_wrap = eq(outputPtr, UInt<2>("h02")) @[Counter.scala 72:24]
      node _outputPtr_wrap_value_T = add(outputPtr, UInt<1>("h01")) @[Counter.scala 76:24]
      node _outputPtr_wrap_value_T_1 = tail(_outputPtr_wrap_value_T, 1) @[Counter.scala 76:24]
      outputPtr <= _outputPtr_wrap_value_T_1 @[Counter.scala 76:15]
[... 258 lines cut ...]
      "org.scalatest" %% "scalatest" % "3.2.9" % "Test", // the capital T is apparently super important???
      "org.rogach" %% "scallop" % "4.0.3",
      "com.lihaoyi" %% "sourcecode" % "0.2.7"
    ),
    scalacOptions ++= Seq(
      "-Xsource:2.11",
      "-language:reflectiveCalls",
      "-deprecation",
      "-feature",
      "-Xcheckinit",
      // Enables autoclonetype2 in 3.4.x (on by default in 3.5)
      "-P:chiselplugin:useBundlePlugin"
    ),
    addCompilerPlugin(("edu.berkeley.cs" % "chisel3-plugin" % "3.4.3").cross(CrossVersion.full)),
    addCompilerPlugin(("org.scalamacros" % "paradise" % "2.1.1").cross(CrossVersion.full))
  )
```

not sure if it matters but jvm is 

```
java 14.0.1 2020-04-14
Java(TM) SE Runtime Environment (build 14.0.1+7)
Java HotSpot(TM) 64-Bit Server VM (build 14.0.1+7, mixed mode, sharing)
```

seldridge, 2021-06-23:
Thanks for reporting this!

I checked and this is _not_ failing on `1.4.2`, but is failing on `1.4.3` and `1.5-SNAPSHOT` (i.e., `master` branch). So, some regression since `1.4.2`. 

#2129 shows the exact same phenotype and may be provide a smaller test case.

seldridge, 2021-06-23:
And to clarify, you're not doing anything wrong here, this is a FIRRTL bug. 😬

makslevental, 2021-06-23:
@seldridge thanks for the quick response. i'll try to rollback to 1.4.2 to get this to work (though i'm betting that'll break something else lol) but looking at https://github.com/chipsalliance/firrtl/issues/2203 am i correct in my reading that i'm violating that constraint on `%` and that that's the source of the issue?

seldridge, 2021-06-23:
Your Chisel code isn't doing anything wrong here as far as I can tell.

FIRRTL IR doesn't have type information on nodes in it's textual representation (except for on ports). However, this necessarily exists in the in-memory representation in the compiler. In the compiler this works by conservatively assigning anything that it can't trivially figure out as `UnknownType`. There is then a pass, `InferTypes`, which populates the type. 

A bug was introduce after 1.4.2 that is causing some pass to generate an `UnknownType`. This can be fixed by either fixing the pass to not do this or adding an "invalidation" to that pass on `InferTypes`. The compiler will then know that that pass "invalidates" `InferTypes` (translation: the pass may generate `UnknownType`) and the compiler needs to re-run `InferTypes` if any downstream pass states that it needs type information.

The Verilog emitter is then barfing when it sees an `UnknownType`.

tl;dr: The Scala FIRRTL compiler treats type information like an analysis and a pass isn't stating that it invalidates type information.