chipsalliance/firrtl #2293: Multi-dimension IO connection bug

sequencer, 2021-07-12:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
```
Seq(firrtl.passes.InferTypes, firrtl.passes.CheckFlows).foldLeft(firrtl.CircuitState(firrtl.Parser.parse(
  """|circuit Top :
     |  module M0 :
     |    output o : { valid : UInt<1>, bits : UInt<1>}[2][2]
     |    skip
     |  module M1 :
     |    output i : { valid : UInt<1>, bits : UInt<1>}[2][2]
     |    skip
     |  module Top :
     |    inst m0 of M0
     |    inst m1 of M1
     |    m1.i[0][0].bits <= m0.o[0][0].bits
     |""".stripMargin), firrtl.UnknownForm)){case (l,r) => r.runTransform(l) }
```
failed with
```
firrtl.passes.CheckFlows$WrongFlow: : [module Top]  Expression m1.i[0][0].bits is used as a SinkFlow but can only be used as a SourceFlow.
```

### What is the expected behavior?
`InferType` pass

### Steps to Reproduce
See scastie below.

### Your environment
FIRRTL 1.4+
### External Information
Report by user in chipsalliance/chisel3#2038

seldridge, 2021-07-12:
This doesn't look like a bug to me. The LHS is a sink. I think the problem here is in the Chisel generation related to `Valid` and the internal `Output` in `Valid`. 

Put differently, CIRCT will produce the same error here:

```
# firtool Tmp.fir               
Tmp.fir:11:21: error: 'firrtl.connect' op has invalid flow: the left-hand-side has source flow (expected sink or duplex flow).
    m1.i[0][0].bits <= m0.o[0][0].bits
                    ^
Tmp.fir:11:21: note: see current operation: "firrtl.connect"(%2, %5) : (!firrtl.uint<1>, !firrtl.uint<1>) -> ()
Tmp.fir:11:16: note: the left-hand-side was defined here.
    m1.i[0][0].bits <= m0.o[0][0].bits
```

sequencer, 2021-07-12:
Got it! Seems a Chisel bug. Sorry, that's my bad!

closed 2021-07-12 by sequencer