chipsalliance/firrtl #2436: (_ BitVec 1) and Bool do not match

liuyic00, 2021-12-08:
I try to use `chiseltest.formal.Formal.verify`  from [chisel-tester2](https://github.com/ucb-bar/chisel-testers2).  
The error says `java.lang.RuntimeException: z3-smtlib encountered an error: "line 7513 column 181: domain sort (_ BitVec 1) and parameter Bool do not match"`  
Then I tried run smt2 emitted from low-firrtl in the z3solver directly, it does show that the smt2 has some problem.

```
(error "line 40370 column 182: domain sort (_ BitVec 1) and parameter Bool do not match")
(error "line 40374 column 182: domain sort (_ BitVec 1) and parameter Bool do not match")
(error "line 40378 column 182: domain sort (_ BitVec 1) and parameter Bool do not match")
(error "line 40382 column 182: domain sort (_ BitVec 1) and parameter Bool do not match")
(error "line 40492 column 110: unknown constant itlb_mdTLB_tlbmd_0.MPORT.data_f (NutCore_s) ")
(error "line 40495 column 110: unknown constant itlb_mdTLB_tlbmd_1.MPORT.data_f (NutCore_s) ")
(error "line 40498 column 110: unknown constant itlb_mdTLB_tlbmd_2.MPORT.data_f (NutCore_s) ")
(error "line 40501 column 110: unknown constant itlb_mdTLB_tlbmd_3.MPORT.data_f (NutCore_s) ")
(error "line 40510 column 121: unknown constant itlb_mdTLB_io_tlbmd_0_f (NutCore_s) ")
(error "line 40513 column 121: unknown constant itlb_mdTLB_io_tlbmd_1_f (NutCore_s) ")
(error "line 40516 column 121: unknown constant itlb_mdTLB_io_tlbmd_2_f (NutCore_s) ")
(error "line 40519 column 121: unknown constant itlb_mdTLB_io_tlbmd_3_f (NutCore_s) ")
(error "line 59215 column 81: unknown constant itlb__GEN_0_f (NutCore_s) ")
(error "line 59216 column 81: unknown constant itlb__GEN_1_f (NutCore_s) ")
(error "line 59217 column 81: unknown constant itlb__GEN_2_f (NutCore_s) ")
(error "line 59218 column 81: unknown constant itlb__GEN_3_f (NutCore_s) ")
(error "line 59799 column 40522: unknown constant itlb_r_0_next (NutCore_s) ")
```

Need a `Bool`, but putted in a `(_ BitVec 1)`

### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

In the smt emitted from low-firrtl:

```smt
...
(define-fun itlb_mdTLB_tlbmd_0.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_0_f state) (ite (itlb_mdTLB_tlbmd_0.MPORT.addr_f state) (_ bv1 1) (_ bv0 1))))
...
(define-fun itlb_mdTLB_tlbmd_1.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_1_f state) (ite (itlb_mdTLB_tlbmd_1.MPORT.addr_f state) (_ bv1 1) (_ bv0 1))))
...
(define-fun itlb_mdTLB_tlbmd_2.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_2_f state) (ite (itlb_mdTLB_tlbmd_2.MPORT.addr_f state) (_ bv1 1) (_ bv0 1))))
...
(define-fun itlb_mdTLB_tlbmd_3.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_3_f state) (ite (itlb_mdTLB_tlbmd_3.MPORT.addr_f state) (_ bv1 1) (_ bv0 1))))
```

### What is the expected behavior?

In the end of lines, use Bool `true false`, not (_ BitVec 1) `(_ bv1 1) (_ bv0 1)` (this solved my problem)
Or other reasonable solutions.

```smt2
(define-fun itlb_mdTLB_tlbmd_0.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_0_f state) (ite (itlb_mdTLB_tlbmd_0.MPORT.addr_f state) true false)))
...
(define-fun itlb_mdTLB_tlbmd_1.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_1_f state) (ite (itlb_mdTLB_tlbmd_1.MPORT.addr_f state) true false)))
...
(define-fun itlb_mdTLB_tlbmd_2.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_2_f state) (ite (itlb_mdTLB_tlbmd_2.MPORT.addr_f state) true false)))
...
(define-fun itlb_mdTLB_tlbmd_3.MPORT.data_f ((state NutCore_s)) (_ BitVec 121) (select (itlb_mdTLB_tlbmd_3_f state) (ite (itlb_mdTLB_tlbmd_3.MPORT.addr_f state) true false)))
```

### Steps to Reproduce

You can get the low-firrtl here [NutCore.txt](https://github.com/chipsalliance/firrtl/files/7676672/NutCore.txt)

- emit the smt `NutCore.smt2`
  - `firrtl -i NutCore.txt -fil NutCore -E smt2
- run smt in z3solver
  - `z3 -smt2 NutCore.smt2`

Then z3 will report some error

Or the full project at [this commit](https://github.com/liuyic00/NutShell/commit/59ef4ca7f02a025a455947c0d450afdf9c80b911)

- run the test
  - `mill __.test`

The test will be failed and report some error

### Your environment

- Chisel Verions: 3.5-SNAPSHOT
- OS: Darwin Kernel Version 21.1.0: Wed Oct 13 17:33:01 PDT 2021; root:xnu-8019.41.5~1/RELEASE_ARM64_T6000 arm64

### External Information

no info

ekiwi, 2021-12-09:
Thanks for the great bug report! I will try to investigate next week.

ekiwi, 2021-12-17:
I hope https://github.com/chipsalliance/firrtl/pull/2446 fixed your issue. If you still see the same problem even with the latest snapshot release, feel free to re-open this issue.

closed 2021-12-17 by ekiwi