chipsalliance/firrtl #2439: Truncating the result of signed addition

youngar, 2021-12-09:
### Checklist

- [ ] Did you specify the current behavior?
- [ ] Did you specify the expected behavior?
- [ ] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [ ] Did you specify relevant external information?

### What is the current behavior?
```
circuit Ops:
  module Ops:
    input sel: UInt<4>
    input is: SInt<8>
    output os: SInt<9>

    os <= SInt(0)

    when eq(sel, UInt(0)):
      os <= add(is, is)
```
 produces:
 ```verilog
module ops(
  input  [3:0] sel,
  input  [7:0] is,
  output [8:0] os
);
  assign os = sel == 4'h0 ? $signed($signed(is) + $signed(is)) : $signed(9'sh0); // @[]
endmodule
```
 
The expression `$signed($signed(is) + $signed(is))` performs the addition at 8 bits instead of 9.

The following two lines of verilog are not equivalent:
```verilog
  assign os = $signed($signed(is) + $signed(is))
  assign os = $signed(is) + $signed(is);
```

---

See https://www.edaplayground.com/x/jbLq to play around with this.

```verilog
module Foo(
  input        sel,
  input  [3:0] is,
  output [4:0] out1,
  output [4:0] out2
);
  assign out1 = $signed(is) + $signed(is);
  assign out2 = sel ? $signed($signed(is) + $signed(is)) : $signed(9'sh0);
endmodule
```
```
    for (int i = 0; i < 4'hf; ++i) begin
    is = i;
    #1;
    $display("%x + %x = %x vs. %x", is, is, out1, out2);
    end
```

```
Compiler version Q-2020.03-SP1-1; Runtime version Q-2020.03-SP1-1;  Dec  9 16:17 2021
00, 00
02, 02
04, 04
06, 06
08, 18
0a, 1a
0c, 1c
0e, 1e
10, 00
12, 02
14, 04
16, 06
18, 18
1a, 1a
1c, 1c
           V C S   S i m u l a t i o n   R e p o r t 
Time: 15 ns
```
---

When the addition is not used in another expression, the outer `$signed()` cast is not present:
```
circuit Ops:
  module Ops:
    input sel: UInt<4>
    input is: SInt<8>
    output os: SInt<9>

    os <= add(is, is)
```
produces:
```
module Ops(
  input  [3:0] sel,
  input  [7:0] is,
  output [8:0] os
);
  assign os = $signed(is) + $signed(is);
endmodule
```
### What is the expected behavior?

The addition should be performed at 9 bits.

### Steps to Reproduce

Above.

### Your environment

- Chisel Verions: 
- OS: 
- Verilator version: 

Fresh compiled FIRRTL.

### External Information

jackkoenig, 2021-12-09:
> A simple fix is probably to avoid forming a compound expression (i.e. avoid emitting the mux and adder in the same Verilog statement) when performing widening arithmetic.

This is the fix that came to mind for me as well. It's a bit surprising, but we've never actually been splitting compound expressions when the outer expression is a `Mux`: https://github.com/chipsalliance/firrtl/blob/4347f57b7ef201251fb2cd463124f2d7cea369f8/src/main/scala/firrtl/passes/SplitExpressions.scala#L55
If I add Mux there, it fixes this issue but I'm a bit worried about a large negative impact on the emitted Verilog. I could make a very focused bug fix splitting out signed add.

Another idea, it's not clear why we wrap the arguments to `mux` in `$signed`, is it ever the case that we need to do that? https://github.com/chipsalliance/firrtl/blob/4347f57b7ef201251fb2cd463124f2d7cea369f8/src/main/scala/firrtl/backends/verilog/VerilogEmitter.scala#L182
Our tests pass if I remove those casts, but such a change is a little scary.

seldridge, 2021-12-17:
I did some digging on this... This bug is not reachable from Chisel code as far as I can tell. So this is clearly a problem, but it's not a showstopping problem. This also likely indicates why a `0.1-BETA` bug was able to avoid detection for 5 years.

To get close to this, you need to be using expanding addition `+&` _and you have to do something custom in a FIRRTL compiler to inline the addition into the mux_. SFC has historically _never_ inlined expressions, except in very limited situations like casts and boolean operations (and this support came very late). The saving grace here is that Chisel emits all expressions as nodes, so they will never be inlined unless a transform moves them.

This Scastie snippet should help get the point across: https://scastie.scala-lang.org/m0nRLBZcTIu7OpXAA6JkLw

This does the signed addition with a temporary wire of width one larger than the addition (which is what you need) and it is not inlined.

I'm mentally downgrading this bug from "showstopping nightmare fuel" to "SFC is weird". The fix proposed in #2440 (extended to all signed arithmetic operations that involve expanding widths) is fine, but this doesn't need to block the release.

PR #2440 (2021-12-18): Fix width of signed addition when input to mux
Extremely focused fix for #2439. I'm not sure if this is the best fix but I think the test is pretty good at least so wanted to put this out there.

Fixes #2439

h/t to @youngar for finding an OG bug.

I'm wondering if it would be better to just remove the casts in the VerilogEmitter, do we ever actually need to cast the inputs to a mux? https://github.com/chipsalliance/firrtl/blob/4347f57b7ef201251fb2cd463124f2d7cea369f8/src/main/scala/firrtl/backends/verilog/VerilogEmitter.scala#L182
That's just a lot scarier of a change 😨 

I'm backporting this [... cut]
closed 2021-12-18 by seldridge