chipsalliance/firrtl #2480: Failed to emit Verilog when using the modulus operator.

gsegura96, 2022-02-08:
### Checklist

- [X] Did you specify the current behavior?
- [X] Did you specify the expected behavior?
- [X] Did you provide a code example showing the problem?
- [X] Did you describe your environment?
- [X] Did you specify relevant external information?

### What is the current behavior?
I get this error on Chisel when trying to emit the Verilog code of a module that uses the modulus operator inside a switch operator.

```
firrtl.FirrtlInternalException: Internal Error! trying to emit unsupported operator: UnknownType
Please file an issue at https://github.com/ucb-bar/firrtl/issues
	at firrtl.Utils$.error(Utils.scala:471)
	at firrtl.Utils$.throwInternalError(Utils.scala:175)
	at firrtl.VerilogEmitter.emitCol(VerilogEmitter.scala:221)
	at firrtl.VerilogEmitter.$anonfun$emitCol$2(VerilogEmitter.scala:213)
	at firrtl.VerilogEmitter.$anonfun$emitCol$2$adapted(VerilogEmitter.scala:212)
        ...
```

Chisel code for the module:
```
import chisel3._
import chisel3.util._
import chisel3.stage.ChiselStage

class ALU_test(width: Int) extends Module {
  val io = IO(new Bundle {
    val op = Input(
      UInt(8.W)
    )
    val in1 = Input(SInt(width.W)) 
    val in2 = Input(SInt(width.W)) 
    val res = Output(SInt(width.W))
  })
  io.res := 0.S(width.W)
  switch(io.op) {
    is(0.U) { io.res := io.in1 + io.in2 }
    is(1.U) { io.res := io.in1 - io.in2 }
    is(2.U) { io.res := io.in1 % io.in2 } 
  }
}
```

### What is the expected behavior?
Before updating my project to Chisel 3.5.0 the Verilog code was generated without a problem. 

### Steps to Reproduce
This Scastie shows the complete error log:
https://scastie.scala-lang.org/4dIFyeXeQPiisTZrLLI9JQ

Apparently FIRRTL can't infer the width of the result. My quick fix was to replace `io.in1 % io.in2` for `0.S(64.W)+(io.in1%io.in2 )` as it forces the width to be 64 and the program emits the Verilog code correctly .

### Your environment
- Chisel Version:  3.5.0
- FIRRTL Version: 1.5.0
- OS: Ubuntu 20.04

### External Information
Might be related to: #2129

ekiwi, 2022-02-08:
firrtl for a regression test:

```
circuit ALU_test :
  module ALU_test :
    input clock : Clock
    input reset : UInt<1>
    output io : { flip op : UInt<8>, flip in1 : SInt<64>, flip in2 : SInt<64>, res : SInt<64>}

    io.res <= asSInt(UInt<64>("h0")) @[main.scala 16:10]
    node _T = eq(UInt<1>("h0"), io.op) @[main.scala 17:17]
    when _T : @[main.scala 17:17]
      node _io_res_T = add(io.in1, io.in2) @[main.scala 18:32]
      node _io_res_T_1 = tail(_io_res_T, 1) @[main.scala 18:32]
      node _io_res_T_2 = asSInt(_io_res_T_1) @[main.scala 18:32]
      io.res <= _io_res_T_2 @[main.scala 18:22]
    else :
      node _T_1 = eq(UInt<1>("h1"), io.op) @[main.scala 17:17]
      when _T_1 : @[main.scala 17:17]
        node _io_res_T_3 = sub(io.in1, io.in2) @[main.scala 19:32]
        node _io_res_T_4 = tail(_io_res_T_3, 1) @[main.scala 19:32]
        node _io_res_T_5 = asSInt(_io_res_T_4) @[main.scala 19:32]
        io.res <= _io_res_T_5 @[main.scala 19:22]
      else :
        node _T_2 = eq(UInt<2>("h2"), io.op) @[main.scala 17:17]
        when _T_2 : @[main.scala 17:17]
          node _io_res_T_6 = rem(io.in1, io.in2) @[main.scala 20:32]
          io.res <= _io_res_T_6 @[main.scala 20:22]
```

seldridge, 2022-02-09:
Yeah, I think this is a dupe of #2129. 😬

jackkoenig, 2022-02-09:
I think https://github.com/chipsalliance/firrtl/issues/2287 is also related.

I looked into this a little bit last night and the issue is that `VerilogModulusCleanup` is introducing a bit selection operation (which changes the type to UInt) without including a cast. I was looking in to fixing it but I actually think that `VerilogModulusCleanup` is actually just unnecessary now due to other changes we've made to `PadWidths` which ensures that the arguments to a `rem` operator are of the same width.

PR #2485 (2022-03-02): Fold VerilogModulusCleanup into LegalizeVerilog
This fixes handling of signed modulus and removes some redundant work.

Fixes #2480, Fixes #2129, Fixes #2287, Fixes https://github.com/chipsalliance/chisel3/issues/2012

This has been a bug for a long time. I've marked it 1.4.x but even that might be a pain to merge due to conflicts with https://github.com/chipsalliance/firrtl/pull/2304

### Contributor Checklist

- [x] Did you add Scaladoc to every public function/method?
- [NA] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you add at least one test demonstrati [... cut]
closed 2022-03-02 by jackkoenig