chipsalliance/firrtl #2490: WDefInstanceConnector support in InstanceKeyGraph

Phantom1003, 2022-03-11:
### What is the current behavior?
I want to write a transform to collect all the register generated
At first, my transform is under VerilogEmitter:
```
 firrtl.stage.transforms.Compiler
  ├── firrtl.passes.ExpandConnects$
  ├── firrtl.passes.ZeroLengthVecs$
  ├── firrtl.stage.TransformManager
  ...
  ├── firrtl.transforms.formal.RemoveVerificationStatements
  ├── firrtl.stage.TransformManager
  │   └── firrtl.passes.ResolveKinds$
  ├── firrtl.passes.CommonSubexpressionElimination$
  ├── firrtl.VerilogEmitter
  │   ├── firrtl.backends.verilog.LegalizeVerilog$
  │   ├── firrtl.passes.memlib.VerilogMemDelays$
  │   ├── firrtl.passes.ResolveFlows$
  │   ├── firrtl.passes.SplitExpressions$
  │   ├── firrtl.passes.ResolveKinds$
  │   ├── firrtl.transforms.CombineCats
  │   ├── firrtl.transforms.InlineBooleanExpressions
  │   ├── firrtl.transforms.LegalizeAndReductionsTransform
  │   ├── firrtl.transforms.DeadCodeElimination
  │   ├── firrtl.transforms.BlackBoxSourceHelper
  │   ├── firrtl.transforms.FixAddingNegativeLiterals
  │   ├── firrtl.transforms.ReplaceTruncatingArithmetic
  │   ├── firrtl.transforms.InlineBitExtractionsTransform
  │   ├── firrtl.transforms.PropagatePresetAnnotations
  │   ├── firrtl.transforms.InlineAcrossCastsTransform
  │   ├── firrtl.transforms.LegalizeClocksAndAsyncResetsTransform
  │   ├── firrtl.transforms.FlattenRegUpdate
  │   ├── firrtl.transforms.DeadCodeElimination
  │   ├── firrtl.passes.VerilogModulusCleanup$
  │   ├── firrtl.transforms.VerilogRename
  │   ├── firrtl.passes.InferTypes$
  │   ├── firrtl.passes.VerilogPrep$
  │   └── firrtl.AddDescriptionNodes
  └── mytransform.RegisterRecord      <<<<  Here  <<<<
```
But, when I compare the .fir and .v, I found that some register is optimized during VerilogEmitter, and it seems like the circuitstate doesn't pass to my custom pass.

So I move my transform under VerilogPrep. Since I use InstanceKeyGraph to generate module path, I got the following error:
```
[error] (run-main-0) firrtl.CustomTransformException:
[error] firrtl.CustomTransformException:
[error]         at firrtl.stage.transforms.CatchCustomTransformExceptions.execute(CatchCustomTransformExceptions.scala:13)
[error]         at firrtl.Transform.transform(Compiler.scala:280)
[error]         at firrtl.Transform.transform$(Compiler.scala:280)
[error]         at firrtl.stage.transforms.CatchCustomTransformExceptions.transform(CatchCustomTransformExceptions.scala:7)
[error]         at firrtl.stage.transforms.UpdateAnnotations.internalTransform(UpdateAnnotations.scala:22)
[error]         at firrtl.stage.transforms.UpdateAnnotations.internalTransform(UpdateAnnotations.scala:8)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.stage.transforms.UpdateAnnotations.transform(UpdateAnnotations.scala:8)
[error]         at firrtl.options.DependencyManager.$anonfun$transform$5(DependencyManager.scala:280)
[error]         at firrtl.Utils$.time(Utils.scala:181)
[error]         at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:280)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.DependencyManager.transform(DependencyManager.scala:269)
[error]         at firrtl.options.DependencyManager.transform$(DependencyManager.scala:255)
[error]         at firrtl.stage.TransformManager.transform(TransformManager.scala:14)
[error]         at firrtl.stage.phases.Compiler.$anonfun$internalTransform$6(Compiler.scala:138)
[error]         at firrtl.Utils$.time(Utils.scala:181)
[error]         at firrtl.stage.phases.Compiler.f$1(Compiler.scala:138)
[error]         at firrtl.stage.phases.Compiler.$anonfun$internalTransform$8(Compiler.scala:143)
[error]         at scala.collection.TraversableLike.$anonfun$map$1(TraversableLike.scala:238)
[error]         at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error]         at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[... 127 lines cut ...]
[error]         at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Translator.transform(Phase.scala:248)
[error]         at firrtl.options.Translator.transform$(Phase.scala:248)
[error]         at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)
[error]         at firrtl.options.Stage.$anonfun$transform$5(Stage.scala:47)
[error]         at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error]         at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122
[error]         at scala.collection.immutable.List.foldLeft(List.scala:89)
[error]         at firrtl.options.Stage.$anonfun$transform$3(Stage.scala:47)
[error]         at logger.Logger$.$anonfun$makeScope$2(Logger.scala:137)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error]         at logger.Logger$.makeScope(Logger.scala:135)
[error]         at firrtl.options.Stage.transform(Stage.scala:47)
[error]         at firrtl.options.Stage.execute(Stage.scala:58)
[error]         at firrtl.options.StageMain.main(Stage.scala:71)
[error]         at starship.utils.stage.Generator.main(Stage.scala)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error] stack trace is suppressed; run last Compile / bgRunMain for the full output
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
```
I do not find a detailed description about the WDefInstanceConnector, why did the InstanceKeyGraph throw this exception?

ekiwi, 2022-03-11:
To make sure that your transform runs before the emitter, add something like:

```.scala
verride def optionalPrerequisiteOf = Forms.LowEmitters
```

That line specifies that your transform needs to be run _before_ any emitter that will run.

About the registers that are added in the VerilogEmitter: This should only happen if you have memories with read latency greater 0 (`SyncReadMem` in chisel). The registers that are added are there by the `VerilogMemDelay` pass to simulate the delayed read behavior. However, in a real tape out, they would actually be part of the SRAM block. Do you still want to count them?

Phantom1003, 2022-03-12:
The register difference is some register has been optimized in Verilog, though I put my transform behind VerilogEmitter, there is still some dead code. I find some registers defined in FIRRTL, but optimized in Verilog. 
I suppose all DeadCodeElimination has been applied on the state return from VerilogEmitter. I also try put the transform under DeadCodeElimination, it did not work. It is until when I put my transform under VerilogPrep, I get the above error. It is really strange.
After putting my transform under VerilogRename, it seems to work well the register list in FIRRTL and Verilog is the same.
BTW, how to read the transform order graph?
```
  ├── firrtl.VerilogEmitter
  │   ├── firrtl.backends.verilog.LegalizeVerilog$
  │   ...
  │   └── firrtl.AddDescriptionNodes
  └── mytransform.RegisterRecord  
```
Does this mean the state will perform in:
```
start 
    -> firrtl.VerilogEmitter 
    -> firrtl.backends.verilog.LegalizeVerilog$ 
    -> firrtl.AddDescriptionNodes 
    -> mytransform.RegisterRecord
```
or
```
start 
    -> firrtl.VerilogEmitter 
    -> firrtl.backends.verilog.LegalizeVerilog$ 
    -> firrtl.AddDescriptionNodes 
start 
    -> mytransform.RegisterRecord
```

I think I do not meet the extra added registers, I use the ReplSeqMem transform to convert those registers to  ExtModule.

closed 2022-11-05 by Phantom1003