chipsalliance/firrtl #2497: Annotation deserialization should not rely on "class" coming first

fabianschuiki, 2022-03-18:
The annotation deserialization in `JsonProtocol` assumes that the "class" field always appears first in a JSON object:
https://github.com/chipsalliance/firrtl/blob/8d2262c0566c5cc7fc66dec7d3e6d1f8102879b5/src/main/scala/firrtl/annotations/JsonProtocol.scala#L312-L325

This assumption breaks easily when the JSON file has been modified/processed/generated by an external tool. Looking for "class" in all fields of the `JObject` instead of pattern matching for the first field would make annotation deserialization a lot more robust.

seldridge, 2022-03-18:
I agree that this is a bug. JSON dictionary order is unordered, so relying on this being the first entry is error prone (though apparently is how Json4s handles type hints so we've never hit this before).

For more info: in MLIR, dictionaries are stored in sorted order and this sorted order can (legally) leak to serialized JSON.

jackkoenig, 2022-03-18:
It should be fairly easy to update line 318 to support `class` in any field. The question is, will json4s deserialization still work even if we collect the type hints correctly? If so, then this is a super easy fix, if not we have a bit of a problem.

PR #2501 (2022-03-25): Fix anno deserialization when class field is not first
Update `findTypeHints` to allow for the "class" field in JSON objects to appear anywhere in the object. This used to rely on the field being the very first in the object, which is easily violated when reading JSON data generated externally, since an object's order of fields is unspecified and can be arbitrarily scrambled.

Fixes #2497.

### Contributor Checklist

- [x] ~Did you add Scaladoc to every public function/method?~
- [x] ~Did you update the FIRRTL spec to include every new feature/behavior?~
- [x] Did you add at least one test dem [... cut]
commit a4aef0f6 (2022-03-25): Fix anno deserialization when class field is not first (#2501) (#2503)

Update `findTypeHints` to allow for the "class" field in JSON objects to
appear anywhere in the object. This used to rely on the field being the
very first in the object, which is easily violated when reading JSON
data generated externally, since an object's order of fields is
unspecified and can be arbitrarily scrambled.

Fixes #2497.

(cherry picked from commit a2d48a5896335a567ddb21ed87fe38cd65d4764d)
commit ec648823 (2022-03-25): Fix anno deserialization when class field is not first (backport #2501) (#2502)

* Fix anno deserialization when class field is not first (#2501)

Update `findTypeHints` to allow for the "class" field in JSON objects to
appear anywhere in the object. This used to rely on the field being the
very first in the object, which is easily violated when reading JSON
data generated externally, since an object's order of fields is
unspecified and can be arbitrarily scrambled.

Fixes #2497.

(cherry picked from commit a2d48a5896335a567ddb21ed87fe38cd65d4764d)

# Conflicts:
#	src/main/scala/firrtl/annotat [... cut]
closed 2022-03-25 by mergify[bot]