chipsalliance/firrtl #2516: Incorrect optimization of a register with a reset

youngar, 2022-04-18:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [ ] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?

```
circuit Test:
  module Test:
    input clock : Clock
    input reset : UInt<1>
    input p: UInt<1>
    output o : UInt<8>
    
    reg r : UInt<8>, clock with :
      reset => (p, UInt<8>(3))
    r is invalid

    o <= r
```
 outputs
 ```
 module Test(
  input        clock,
  input        reset,
  input        p,
  output [7:0] o
);
  assign o = 8'h0;
endmodule
```
### What is the expected behavior?

I expected the `r is invalid` to be equivalent to `r <= UInt<8>(0)`:

```
module Test(
  input        clock,
  input        reset,
  input        p,
  output [7:0] o
);
  reg [7:0] r;
  assign o = r;
  always @(posedge clock) begin
    if (p) begin
      r <= 8'h3;
    end else begin
      r <= 8'h0;
    end
  end
endmodule
```

### Steps to Reproduce

### Your environment

- Chisel Verions: 
- OS: 
- Verilator version: 

### External Information

@seldridge has indicated that the InferResets pass is missing a case to handle `IsInvalid`:
https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/transforms/RemoveReset.scala#L56

[This](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/transforms/RemoveReset.scala#L77) match is going to move the reset value into the resets map. However, the pass never comes back to it because there is no case to match for `IsInvalid`.

ekiwi, 2022-04-19:
It seems to me like this would be a valid output:

```
module Test(
 input        clock,
 input        reset,
 input        p,
 output [7:0] o
);
 assign o = 8'h3;
endmodule
```

Since invalid could be any constant and thus it should be valid - in the original code - to replace `r is invalid` with `r <= UInt(3)` which then would mean the register is stuck at `UInt(3)` and thus can be replaced by a constant `UInt(3)`.

jackkoenig, 2022-04-19:
I agree that it should be legal to replace it with constant `UInt(3)`, the problem is that it is actually replacing it with constant `UInt(0)` which is a bug.

ekiwi, 2022-04-19:
> I agree that it should be legal to replace it with constant `UInt(3)`, the problem is that it is actually replacing it with constant `UInt(0)` which is a bug.

Yes. It is a bug. I was just wondering if there was a way to still eliminate the register and was curious to see if others agreed.

jackkoenig, 2022-04-19:
Ah yes, I agree with you. This should give the same output as
```
input clock : Clock
input reset : UInt<1>
input p: UInt<1>
output o : UInt<8>
                                                                                                                                                                                                                          
reg r : UInt<8>, clock
r  <= UInt<8>(3)

o <= r
```

Which does indeed give:
```verilog
module Test(                                                                                                                                                                                                                  
  input        clock,
  input        reset,
  input        p,
  output [7:0] o
);
  assign o = 8'h3;
endmodule
```

seldridge, 2022-04-19:
I think there are two things that are important to disentangle:

1. What should `is invalid` be defined as in the FIRRTL specification?
2. What does `is invalid` actually mean in the SFC?

The difficulty here is that registers matter a lot for formal equivalence and any deviation from SFC behavior, even in situations with undefined behavior, can result in formal equivalence failures if the optimization choice around an invalid results in removal of a register.

If (1) were better defined, then `assign o = 8'h3` could be a legal optimization. However, that really needs a definition of invalid value (which is totally aligned with your thinking, Kevin!). Due to choices in how the SFC works, this is not actually a safe optimization given (2) if you want to get SFC-compatible output.

Elaborating on (2), `invalid` in the SFC means one of the following, based on context:

  a. An invalid arising from a `validif` means "the value on the true leg of the `validif`" (or `validif` means connect).
  b. An invalid in the default of a `when` block means "the same value as the last value in the equivalent mux tree". (This is the same as case (a)).
  c. A register with an initial value that is invalid _within module scope and looking through only wires and connects_ means "this is actually a reset-less register."
  d. An invalid in any other context means "zero".

Note that (a) and (b) do not actual deal with an invalid _value_, but rely on the use of an `is invalid` statement which means that an invalid node/wire or an invalid module output do not contribute to these interpretations and instead fall into interpretation (d).

The changed proposed here is to get SFC in-line with interpretation (d) by fixing a bug.

This is an extremely long-winded way of saying that SFC semantics around invalid are incredibly, incredibly tedious, and please don't change them or add any more special cases like the brittle (c) to cover this case. 😅 

I am open to changing this in the future, especially inside MFC, but it requires a true notion of invalid values which the SFC was never architected to support. E.g., if it's legal to optimize `assign o <= 8'h3` above, then it should also work in the following case where the invalidation requires a global analysis:

```scala
circuit Foo:
  module Bar:
    output inv: UInt<8>

    inv is invalid

  module Foo:
    input clock : Clock
    input p: UInt<1>
    output o : UInt<8>

    reg r : UInt<8>, clock with :
      reset => (p, UInt<8>(3))

    inst bar of Bar
    r <= bar.inv

    o <= r
```

ekiwi, 2022-04-19:
> However, that really needs a definition of invalid value (which is totally aligned with your thinking, Kevin!).

I just got excited because the optimization I proposed works with both invalid definitions that we have been discussing: _arbitrary, but fixed value_ or _arbitrary value, potentially different value eveytime the same invalid is read_

> E.g., if it's legal to optimize `assign o <= 8'h3` above, then it should also work in the following case where the invalidation requires a global analysis:

Optimizations are always optional.

seldridge, 2022-04-19:
I like your thinking. Have you considered the addition of freeze to give you finer-grained control? The design point I want to play with is [poison](https://llvm.org/docs/LangRef.html#poisonvalues)/[freeze](https://llvm.org/docs/LangRef.html#freeze-instruction) which would have the following semantics:

```scala
wire foo: UInt<8>
foo <= Poison<8>

node x = freeze(foo) // This is a random value
node y = freeze(bar) // This is another random value

assert(eq(x, x)) // always true
assert(eq(x, y)) // sometimes true, sometimes false
```

Poison is then defined for all ops and propagates aggressively. However, Jack has proposed putting a "block" on poison propagation that is either a compile time or run-time error. E.g., something like this would enable poison (uninitialized?) to run around safely, but not escape into someplace it shouldn't be like a program counter due to an unapplied reset:

```scala
reg programCounter: UInt<64>, gatedClock

reg resetVector: UInt<64>, clock with: (reset => (reset, UInt<64>("h800000000")))

programCounter <= unpoisoned(resetVector)
```

This is all naively in the realm of "let's just build a better X". There's years of painful lessons from software compilers. 🥲 

Your notion of "random value every time it is read" is similar to a freeze associated with a clock.

ekiwi, 2022-04-19:
> I like your thinking. Have you considered the addition of freeze to give you finer-grained control?

My approach would be that everything is essentially immediately frozen after the poison is generated:

```
wire foo: UInt<8>
foo <= freeze(Poison<8>)

node x = foo
node y = bar

assert(eq(x, x)) // always true
assert(eq(x, y)) // always true
```

Then we can see if there are reasonable cases where this would inhibit needed optimizations. However, the original firrtl compiler is already more agressive sometimes.

PR #2520 (2022-04-22): Fix optimization of register with reset but invalid connection
Fixes #2516

Previously,
```
reg r : UInt<8>, clock with :
  reset => (p, UInt<8>(3))
r is invalid
```
would compile to:
```
reg r : UInt<8>, clock
r <= UInt<8>(0)
```
now it compiles to:
```
reg r : UInt<8>, clock
r <= UInt<8>(3)
```

### Contributor Checklist

- [NA] Did you add Scaladoc to every public function/method?
- [NA] Did you update the FIRRTL spec to include every new feature/behavior?
- [x] Did you add at least one test demonstrating the PR?
- [x] Did you delete any extraneous printlns/debugging code?
- [x] Did you spe [... cut]
commit 5093da03 (2022-04-22): Fix optimization of register with reset but invalid connection (#2520)

Fixes #2516

Previously,

reg r : UInt<8>, clock with :
  reset => (p, UInt<8>(3))
r is invalid

would compile to:

reg r : UInt<8>, clock
r <= UInt<8>(0)

now it compiles to:

reg r : UInt<8>, clock
wire r_1 : UInt<8>
r_1 is invalid
r <= mux(reset, UInt<8>(3), r_1)

This is consistent with the behavior for a reset with an asynchronous
reset.
commit 09b1f226 (2022-04-22): Fix optimization of register with reset but invalid connection (#2520) (#2523)

Fixes #2516

Previously,

reg r : UInt<8>, clock with :
  reset => (p, UInt<8>(3))
r is invalid

would compile to:

reg r : UInt<8>, clock
r <= UInt<8>(0)

now it compiles to:

reg r : UInt<8>, clock
wire r_1 : UInt<8>
r_1 is invalid
r <= mux(reset, UInt<8>(3), r_1)

This is consistent with the behavior for a reset with an asynchronous
reset.

(cherry picked from commit 5093da03083a37a0a7bdaf44f9867d7f7a0a5980)
commit 486194c4 (2022-04-22): Fix optimization of register with reset but invalid connection (#2520) (#2522)

Fixes #2516

Previously,

reg r : UInt<8>, clock with :
  reset => (p, UInt<8>(3))
r is invalid

would compile to:

reg r : UInt<8>, clock
r <= UInt<8>(0)

now it compiles to:

reg r : UInt<8>, clock
wire r_1 : UInt<8>
r_1 is invalid
r <= mux(reset, UInt<8>(3), r_1)

This is consistent with the behavior for a reset with an asynchronous
reset.

(cherry picked from commit 5093da03083a37a0a7bdaf44f9867d7f7a0a5980)
closed 2022-04-22 by jackkoenig