chipsalliance/firrtl #2539: Unsupported platform: protoc-3.5.1-osx-aarch_64.exe

ilongshan, 2022-08-10:
### Checklist

- [x] Did you specify the current behavior?
- [x] Did you specify the expected behavior?
- [x] Did you provide a code example showing the problem?
- [x] Did you describe your environment?
- [x] Did you specify relevant external information?

### What is the current behavior?
Error occurred while `sbt compile` :
```
protoc-jar: downloading: https://repo.maven.apache.org/maven2/com/github/os72/protoc/3.5.1-build2/protoc-3.5.1-build2-osx-aarch_64.exe
[error] java.lang.RuntimeException: error occurred while compiling protobuf files: Unsupported platform: protoc-3.5.1-osx-aarch_64.exe
[error] 	at sbtprotobuf.ScopedProtobufPlugin.executeProtoc(ProtobufPlugin.scala:112)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.compile(ProtobufPlugin.scala:128)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.$anonfun$sourceGeneratorTask$3(ProtobufPlugin.scala:170)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$1(FileFunction.scala:80)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$4(FileFunction.scala:153)
[error] 	at sbt.util.Difference.apply(Tracked.scala:414)
[error] 	at sbt.util.Difference.apply(Tracked.scala:394)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$3(FileFunction.scala:149)
[error] 	at sbt.util.Difference.apply(Tracked.scala:414)
[error] 	at sbt.util.Difference.apply(Tracked.scala:389)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$2(FileFunction.scala:148)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.$anonfun$sourceGeneratorTask$1(ProtobufPlugin.scala:172)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.Execute.work(Execute.scala:291)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
[error] 	at java.base/java.lang.Thread.run(Thread.java:833)
[error] Caused by: java.io.FileNotFoundException: Unsupported platform: protoc-3.5.1-osx-aarch_64.exe
[error] 	at com.github.os72.protocjar.Protoc.extractProtoc(Protoc.java:223)
[error] 	at com.github.os72.protocjar.Protoc.extractProtoc(Protoc.java:184)
[error] 	at com.github.os72.protocjar.Protoc.runProtoc(Protoc.java:68)
[error] 	at com.github.os72.protocjar.Protoc.runProtoc(Protoc.java:55)
[error] 	at $9fc64a419ca894a87f5e$.$anonfun$protobufSettings$4(build.sbt:70)
[error] 	at $9fc64a419ca894a87f5e$.$anonfun$protobufSettings$4$adapted(build.sbt:70)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.executeProtoc(ProtobufPlugin.scala:110)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.compile(ProtobufPlugin.scala:128)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.$anonfun$sourceGeneratorTask$3(ProtobufPlugin.scala:170)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$1(FileFunction.scala:80)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$4(FileFunction.scala:153)
[error] 	at sbt.util.Difference.apply(Tracked.scala:414)
[error] 	at sbt.util.Difference.apply(Tracked.scala:394)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$3(FileFunction.scala:149)
[error] 	at sbt.util.Difference.apply(Tracked.scala:414)
[error] 	at sbt.util.Difference.apply(Tracked.scala:389)
[error] 	at sbt.util.FileFunction$.$anonfun$cached$2(FileFunction.scala:148)
[error] 	at sbtprotobuf.ScopedProtobufPlugin.$anonfun$sourceGeneratorTask$1(ProtobufPlugin.scala:172)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:62)
[error] 	at sbt.std.Transform$$anon$4.work(Transform.scala:68)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:282)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:23)
[error] 	at sbt.Execute.work(Execute.scala:291)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:282)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:64)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
[error] 	at java.base/java.lang.Thread.run(Thread.java:833)
[error] (Protobuf / protobufGenerate) error occurred while compiling protobuf files: Unsupported platform: protoc-3.5.1-osx-aarch_64.exe
```
### What is the expected behavior?
No error when `sbt comiple`
### Steps to Reproduce
clone the project and run `sbt compile`

### Your environment

- firrtl Verions:  master branch
- OS:  masOS Monterey

arun13e, 2023-04-12:
Any quick fixes or hacks for now to bypass this?

ekiwi, 2023-04-12:
@jackkoenig  Can we just remove the protobuf support?

jackkoenig, 2023-04-12:
Protobuf is removed from firrtl brought in tree to Chisel and this repo is shifted to bug fixes only. Fixes this so we can build this repo going forward may make sense, but supporting ARMv8 MacOS is not a priority for that. That being said, we could potentially fix it by not using https://github.com/os72/protoc-jar any more (we are on the latest version) and instead requiring the user who is building firrtl to install protoc themselves.

jackkoenig, 2023-04-12:
That repo is also Apache 2.0 licensed so we could also just in tree and fix the logic for ARM Macs