chipsalliance/firrtl #2691: firrtl internal error- caused by java

Prasanthyg, 2024-06-12:
ARTIFACTS (.v) files not being generated after running after invoking the command- "rtl -a /demo/arch/pynqz1.tarch -s true" the RTL Summary is generated but not the .v files

OS- Windows 10
Errors:  at firrtl.Utils$.error(Utils.scala:471)
        at firrtl.Utils$.throwInternalError(Utils.scala:175)
        at firrtl.stage.phases.CatchExceptions.transform(CatchExceptions.scala:31)
        at firrtl.stage.phases.CatchExceptions.transform(CatchExceptions.scala:10)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:38)
        at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
        at firrtl.options.Translator.transform(Phase.scala:248)
        at firrtl.options.Translator.transform$(Phase.scala:248)
        at firrtl.options.phases.DeletedWrapper.transform(DeletedWrapper.scala:15)......
..................and a few more lines

Caused by: java.io.FileNotFoundException: /work/bram_dp_128x8192.v (Permission denied)
        at java.base/java.io.FileOutputStream.open0(Native Method)
        at java.base/java.io.FileOutputStream.open(FileOutputStream.java:298)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:237)
        at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:187)
        at java.base/java.io.PrintWriter.<init>(PrintWriter.java:309)
        at firrtl.transforms.BlackBoxSourceHelper$.writeTextToFile(BlackBoxSourceHelper.scala:187)
        at firrtl.transforms.BlackBoxSourceHelper.$anonfun$execute$1(BlackBoxSourceHelper.scala:122)

These are the list of error being generated

Please help!!!

jackkoenig, 2024-06-12:
Can you provide more information? What version of FIRRTL (or presumably Chisel) are you using? How are you invoking it? Do you have a reproducible example?

Prasanthyg, 2024-06-12:
Sorry for vaguely reproducing the error, I'm just very new t:o this environment and only used to vivado etc.., anyway to answer your question I'm simply following the step wise tutorial given in the tensil website to get used to the flow. 

Anyway I'm following the implementation of Resnet on Pynq Z1 board and this error comes up after invoking the command:

tensil rtl -a /demo/arch/pynqz1.tarch -s true

The rtl Resource table is generated in the CLI by the ARTIFACT portion detailing the 3 .v files is not generated and instead these errors are generated in the log. 

Been stuck with this for two days.

jackkoenig, 2024-06-13:
This is more of a question for that project than this repo. Tensil also appears to be on an old version of firrtl--in fact, this entire project is deprecated as Chisel has moved on to a new implementation of the firrtl compiler (https://github.com/llvm/circt). I am willing to assist debugging reduced test cases on the newest version of Chisel that still uses this repo (v3.6.0), but generally I advise all users of Chisel to bump to the newest version (v6.4.0) and use the new compiler.